FreeCAD/FreeCAD #16470: DXF importer (C++): _PyEval_EvalFrameDefault returned a result with an exception set error when importing file

furgo16, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD
2. Create a new document
3. File > Import...
4. Select the previously uncompressed [DXF file attached](https://github.com/user-attachments/files/16959918/Layertest.zip)
5. Use the import settings in the screenshoot below. Most importantly, uncheck `Use legacy Python exporter`, as we want to use the C++ importer
6. Click OK
7. The DXF file entities are imported apparently successfully. The Report view and the notification area popu show the following error:

```
pyException: /snap/freecad/1132/usr/Mod/Draft/importDXF.py(2854)<class 'SystemError'>: _PyEval_EvalFrameDefault returned a result with an exception set
0: _PyEval_EvalFrameDefault returned a result with an exception set
```

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38693 (Git) Snap 1132
Build type: Release
Branch: main
Hash: 85ade21436d4ce618691720688b299c9b6902590
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: /Classic/
Installed mods: 
  * OpenTheme 2024.7.30
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
cc @kpemartin

kpemartin, 2024-09-13:
This sounds like another variant on the same problem as #16068 and #13597, but I'll have a look at it to see if my PR fixes it too.

furgo16, 2024-09-14:
Thank you. The related PR is at https://github.com/FreeCAD/FreeCAD/pull/16511. I will test it as soon as I can build it.

kpemartin, 2024-09-15:
I've had a look and this is indeed a slightly different manifestation of the same as as #16068 and #13597. Note that to see this particular manifestation it was necessary to import the file a second time (by which time the `Part` module had been imported so `Document::addObject` succeeded), and the lingering exception state remained until the document was recalculated. The exception state was noticed in the same questionable manner but this time within `_PyEval_EvalFrameDefault`.