FreeCAD/FreeCAD #16471: PartDesign: Unable to Create Additive Loft from Circle to Rectangle

arena-, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a simple Additive Loft from Circle to Rectangle fails with Input Error: "BRep_API: command not done". 
The same Additive Loft works if done from Rectangle to Circle.

Setup:
[image]

From Rectangle to Circle (works):
[image]

From Circle to Rectangle (fails):
[image]

To reproduce, open the file: 
[loft.zip](https://github.com/user-attachments/files/16973136/loft.zip) (or create two sketches on the same plane, one rectangle and one circle, and add a sketch attachment offset to one of them)

Steps
1. Go to Part Design workbench
2. Click anywhere to deselect anything that may be selected
3. Click Additive Loft button
4. Select Circle profile, Click Ok
5. Click Add Section and select the Rectangle profile
6. Click Ok

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.5.24
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arena-, 2024-09-12:
I just tested it in 0.21.2 and it works both ways in that version. 

Full version info: 

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.5.24
  * sheetmetal 0.4.24
[/code]

Syres916, 2024-09-12:
The fix must have been implemented after your build as it works both ways using:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38698 (Git)
Build type: Release
Branch: main
Hash: 0946c6880def66bc0c611dad37140da3f126a6c3
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * GDML 2.0.0
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66

```
I'm guessing it was this commit https://github.com/FreeCAD/FreeCAD/commit/2dd45265316f2928fbeca0d12b82baa6f667cb97

luzpaz, 2024-09-12:
Closing as Fixed. If you can reproduce on weekly or the next RC release, please re-open.

mcavallerin, 2024-09-15:
Same errore on
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/Dark modern/Qt default
Installed mods: 
  * sheetmetal 0.4.24
  * fasteners 0.5.27

If you pre-select sketches it works, bit this lower the user experience quality

luzpaz, 2024-09-15:
@arena- can you reproduce on latest weekly ?  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

arena-, 2024-09-15:
Just tested, I was able to reproduce the same error on: 

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.47
  * fasteners 0.5.24
  * sheetmetal 0.4.25