FreeCAD/FreeCAD #16473: PartDesign: Adding a feature removes Body transparency

arena-, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a Body is transparent, adding a new feature immediately removes the transparency for the Body.

Body is 50% transparent: 
[image]

Adding a 5mm fillet, transparency is removed:
[image]

Doing the same operation is 0.21.2 it works as expected:
[image]

To reproduce, download this file: 
[transparency.zip](https://github.com/user-attachments/files/16974743/transparency.zip) (or create a simple rectangular pad and then go to click on the Body -> View -> Transparency and set to "50")

1. Click on an edge
2. Click "Add Fillet" 

- Observe how the transparency has been removed

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.5.24
  * sheetmetal 0.4.24

AND the 0.21.2 where it works:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.5.24
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-12:
This issue is basically a demonstration (therefore duplicate of) https://github.com/FreeCAD/FreeCAD/issues/15503

luzpaz, 2024-09-12:
Closing as Duplicate.

closed 2024-09-12 by luzpaz