FreeCAD/FreeCAD #16474: Variable set: Rename property deletes the property

subjektivdk, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are problems when right-clicking to rename, change property

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.12.2
  * DynamicData 2.46.0
  * fasteners 0.4.56
  * Help 1.0.3
  * ThreadProfile 1.87.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-09-12:
Could you add detailed steps to reproduce your issue, along with what you expect and what it actually happens?

I might be wrong, but I don't think it's yet possible to rename a property. The one with the old name needs to be removed first, and the one with the new name then created anew.

benj5378, 2024-09-12:
> The one with the old name needs to be removed first, and the one with the new name then created anew.

Should still be done under the hood

kaiwas, 2024-09-12:
Similar topic. Not related to Varset, but definitely related to editing properties.
https://github.com/FreeCAD/FreeCAD/issues/12837

maxwxyz, 2024-09-12:
@pieterhijma FYI

luzpaz, 2024-10-24:
CC @pieterhijma

pieterhijma, 2024-10-29:
Sorry, I lost track of this one.

In general it is not possible to rename properties as far as I'm aware. The only possibility that comes close to what was reported here is a user right-clicking in the Property view to rename the property **group**. Perhaps @subjektivdk was confused by that and it was perceived as deleting the property. 

Unless we are provided with steps to reproduce within a week or so, I would recommend to close this issue and refer to #16222.

maxwxyz, 2024-10-29:
Closed in favor of #16222

closed 2024-10-29 by maxwxyz