FreeCAD/FreeCAD #16478: Windows launch difficulty of FreeCAD 1.0.0 rc1

Xpendable172, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm having a tough time getting FreeCAD 1.0.0 rc1 to launch without crashing.  I will always see the splash screen, but the FreeCAD UI only successfully displays 10% of the time.  The other 90% of the time, it exits either before the UI is shown, or within 1 second of the UI being shown.  On the 10% of times it stays up, there are lots of access violations being reported in the report view:
"Unhandled Base::Exception caught in GUIApplication::notify.
The error message is:  Access Violation"

### Full version info

```shell
Not possible, difficult to get it to launch without it crashing immediately.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-12:
I've just launched successfully. @Xpendable172, could you provide some information, like, Windows version?

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

luzpaz, 2024-09-12:
Can you run FreeCAD from the terminal and add the `--version --verbose` flags ?

Xpendable172, 2024-09-12:
FreeCAD 1.0.0 Revision: 38642 (Git)

Library versions:
boost    1_82
Coin3D   4.0.2
Eigen3   3.4.0
OCC      7.7.2
Qt       5.15.13
Python   3.11.9
PySide   5.15.8
shiboken 5.15.8
SMESH    9.8.0.2
VTK      9.2.6
xerces-c 3.2.5

Xpendable172, 2024-09-12:
> @Xpendable172 have you tried @kisolre 's build based on the old LibPack available [here](https://github.com/kisolre/FreeCAD/releases/tag/weekly_Win7) (ignore the Windows 7 name, it runs on Windows 10 for definite), also could you advise if you've ever run an older version of FreeCAD on the that particular system if that build crashes also.

Yes, I am successfully able to run 0.21.2 on the same computer.  No, I have not tried the weekly_Win7 build you mentioned above.

Xpendable172, 2024-09-12:
> > Yes, I am successfully able to run 0.21.2 on the same computer.
> 
> Excellent, well the Win7 weekly rc1 build is based on exactly the same libraries as your working 0.21.2 build so my conclusion is there's likely to be a graphics card/driver issue with your system and the Conda build libraries of the standard weekly build.

I can certainly try that.  But why does it _sometimes_ work?

Xpendable172, 2024-09-12:
> Excellent, well the Win7 weekly rc1 build is based on exactly the same libraries as your working 0.21.2 build so my conclusion is there's likely to be a graphics card/driver issue with your system [...]

Ok, I have tried this build, and I am getting the same results.  On first launch, it did launch, but showed an illegal storage access error.  See attachment.  Haven't gotten it to come up since.
[image]

Xpendable172, 2024-09-12:
> I've just launched successfully. @Xpendable172, could you provide some information, like, Windows version?

See above, I was able to catch a launch and get this information and updated the version info in the ticket.  Thanks!

Xpendable172, 2024-09-12:
> OK, now that I've seen the screenshot of the boot log, the root cause is the Render Workbench is not up-to-date. The `killTimer` warning is the give away. Please make sure you keep Addons up-to-date, I've raised two pull requests with Addon owners with specific FreeCAD version 1.x fixes.
> 
> The current version of the Render Wb is `2024.08.04`

Ok, updating them now.  I will report back results.  Thanks!

Xpendable172, 2024-09-12:
Unfortunately, it didn't change anything... FreeCAD 1.0 rc1 still only gets past the splash screen < 10% of the time.  Just to be be transparent on what I did to update the workbenches... I launched 0.21.2 again, went into the AddOn manager, selected the updates available filter (Render, Curves, and Assembly all had updates), updated, relaunched 0.21.2 to make sure all was okay, then closed down 0.21.2 and tried launching FreeCAD 1.0.  Same problem of splash screen going away then crash (nothing on the screen).  Haven't yet gotten it to come back up successfully.

Syres916, 2024-09-12:
OK, one last thing to try before I have to take a step back, with FreeCAD closed down _**rename**_ (don't delete) your two config files:
C:\Users\ _username_\AppData\Roaming\FreeCAD\user.cfg to something like _userold.cfg_
C:\Users\ _username_\AppData\Roaming\FreeCAD\system.cfg to something like _systemold.cfg_
Rename the folder C:\Users\ _username_\AppData\Roaming\FreeCAD\Mod to something like _OldMod_

So now when you start any of 1.0rc1 builds you have no interference from any Addons and there's no corrupt data in your config files.

You can always then rename the _OldMod_ folder back to _Mod_ and rename the config files back to how they were if needed.

Xpendable172, 2024-09-12:
> OK, one last thing to try before I have to take a step back, with FreeCAD closed down _**rename**_ (don't delete) your two config files: C:\Users\ _username_\AppData\Roaming\FreeCAD\user.cfg to [...]

Good idea.  Trying  now.  SUCCESS!  That resolved the issue.

fritzeken, 2024-09-13:
Thank you very much! I had the same problem and solved it the same way.

kisolre, 2024-09-13:
So could any of you try to pinpoint the issue? Restore config files one by one, copy mods from mod backup folder to mod folder and restart FreeCAD after each change?

fritzeken, 2024-09-17:
I renamed user.cfg, system.cfg and folder Mod to _old - the programstarts and runs stable. I haven't worked too much with ORC1 yet and today
1. Files user_old and sytem_old renamed individually: stable.
2. Both files renamed together: stable
3. Folder Mod_old was not changed: stable, the program did not create a new folder Mod.
4. Folder Mod renamed: stable
Conclusion: The program must be forced to recreate user.cfg and system.cfg.

nickdollimount, 2025-02-09:
I got this issue with the public stable release. It appeared to be related to the Fasteners addon. Everything was going fine until I went into the Assembly workbench to add some screw joints. Renaming the Mod folder allowed me to use the existing screw model that was generated with the addon to add a handful of them into the assembly without issue. Not sure if that helps.

luzpaz, 2025-05-20:
Is this still reproducible?

closed 2025-08-05 by github-actions[bot]