FreeCAD/FreeCAD #16479: RC1 Elmer FEM Examples do not work properly

domcivocato, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to do an electrostatics simulation, but i wouldnt get reasonable results. I tried to do the FEM examples, that should be good from the get-go, but they also didnt have reasonable results. 

[image]

The values are around 0 indicating that the boundary conditions do not get passed well and they assume all boundary conditions to be 0, i guess.

In Ondsel 2024.2.1 its working and i recall that in former versions of FreeCAD it also worked. You might check in on it. 

[image]

### Full version info

```shell
OS: TUXEDO OS 3 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
/cc @marioalexis84 @FEA-eng

FEA-eng, 2024-09-12:
Support for Elmer has been broken recently but should be fixed with PRs merged on Monday. Unfortunately, they haven't made it to RC1.

luzpaz, 2024-09-12:
@domcivocato new builds will be generated tomorrow for the weekly build. Please consider downloading them to reproduce? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-09-12:
@FEA-eng which PR(s) is this related to ?
16351 ? 
16394 ?

FEA-eng, 2024-09-12:
> @FEA-eng which PR(s) is this related to ? 16351 ? 16394 ?

Yes, those 2. It works normally for me in the weekly build from around half an hour ago.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

domcivocato, 2024-09-12:
@luzpaz Will download and try tomorrow again with the weekly build.

FEA-eng, 2024-09-22:
@domcivocato Could you check with the newest weekly build and close the issue if it works?

domcivocato, 2024-09-22:
Hey, i wanted to test it. But something came in between. Ill test it at tuesday latest.

Best regards, domcivocato

closed 2024-09-24 by domcivocato