FreeCAD/FreeCAD #16481: Part Workbench Sketch: Creates multiple extrusions

billdoerr, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the sketch tool via the Parts WB multiple extrusions are created when then extruding.

Steps to reproduce:
1. Launch FreeCAD 1.0.0RC1.38642.
2. Select Part Workbench.
3. Click New Document.
4. Click the Sketch icon.
5. Select XY-plane (or any plane).
6. Click the Create Rectangle tool and create a simple rectangle.
7. Click close.
8. Click the extrude tool. Default values are ok to use.
9. Click Apply but don't close dialog.
10. Click the Reverse check box then click Apply.
11. You will then see another rectangle extruded.
12. Click the Symmetric check box and then click Apply.
13. Yet another rectangle is created.
14. Click Ok to close dialog.
15. Click on the Model tab. You will then see multiple bodies created from the result of steps 10 - 13.
16. Expectation is that this would work the same as with the Part Design WB.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * AirPlaneDesign 0.4.0
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * fasteners 0.4.50
  * Help 1.0.3
  * Plot 2022.4.17
  * Ship 2022.4.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MarcusMWilliams, 2024-09-12:
plane

maxwxyz, 2024-09-12:
this is the currently designed workflow in Part (and many other WB e.g. Draft).

closed 2024-09-12 by maxwxyz