FreeCAD/FreeCAD #16482: Measure: If measure task dialog is open pressing 'Enter' while editing spreadsheet doesn't work correctly

luzpaz, 2024-09-12:
There's a second issue here: unwanted behaviour. If the Measure Task dialog is open, pressing enter while editing the spreadsheet won't actually be reached by the spreadsheet but be swallowed by the taskdialog. So if editing a value and pressing enter in the spreadsheet, the value won't be applied, but instead trigger the Measure Task dialog (even though it's not selected/in use). @luzpaz can you open a new issue for this?

_Originally posted by @benj5378 in https://github.com/FreeCAD/FreeCAD/issues/16401#issuecomment-2346461312_

hlorus, 2024-09-17:
@wwmayer The measure dialog currently adds an event filter to check for Enter/Escape key presses. Without the event filter the task dialog looses focus as soon as an element in the 3d view is selected which makes the dialog unresponsive to key presses. Is there a better way to handle this?

wwmayer, 2024-09-24:
> Is there a better way to handle this?

No. Using an event filter is the best way to handle it.

FEA-eng, 2025-10-17:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * A2plus 0.4.68
  * freecad.gears 1.3.0
```

FEA-eng, 2026-11-05:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/9e930b694e25f77beeaab13b0c2e8998f95ca3d4

Return is left to the active editor while editing a cell, then the Measure shortcut becomes available again when focus goes back to the task or 3D view.