FreeCAD/FreeCAD #16484: The MAC ARM64 RC1 download file is damaged

dfadsr, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've tried twice to install it and it replies with the same message.

### Full version info

```shell
https://objects.githubusercontent.com/github-production-release-asset-2e65be/93114989/f345d246-ead7-442b-b13a-ff4956d41bfc?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240912%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240912T172916Z&X-Amz-Expires=300&X-Amz-Signature=0eb7d7a8274c08eb78767244b5b5d3cf51f627e82ceb5e599f829645227ba746&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=93114989&response-content-disposition=attachment%3B%20filename%3DFreeCAD_1.0.0-conda-macOS-arm64-py311.dmg&response-content-type=application%2Foctet-stream
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

diekhans, 2024-09-13:
I have the same issue. I've attached a log file with records matching the string freecad when it fails.  

If I run /Applications/FreeCAD.app/Contents/MacOS/FreeCAD  from the command line, it starts with no problem.

[freecad.log](https://github.com/user-attachments/files/16986779/freecad.log)

PeterKnego, 2024-09-13:
That's a problem with code signing. As a temporary measure you can do this:

`xattr -c /Applications/FreeCAD.app`

luzpaz, 2024-09-14:
Closing as known issue and duplicate.

closed 2024-09-14 by luzpaz

adrianinsaval, 2024-09-14:
I uploaded yesterday a signed version so this issue shouldn't be present on rc1 anymore

dfadsr, 2024-09-14:
> I uploaded yesterday a signed version so this issue shouldn't be present on rc1 anymore

Very well, I will try it. In general it seems Apple is rather strict with their security policies