FreeCAD/FreeCAD #16485: BIM, DRAFT: Crash when trying to use a DXF as a drawing template.

gianlucafiore, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I import a DXF plan to use as a 3D construction base for a building, and with very few movements to create rectangles or walls, or try to make a movement, the program crashes and closes unexpectedly.

[TEST FILE](https://github.com/user-attachments/files/16983842/Edificio.ultimo.2.zip)

STEPS to replicate:

1-Import a DXF already cleaned of dimensions and nomenclatures or annotations. (All layers grouped into a single object, without layers, (and not converting splines and circles into polylines).

2- by default the DXF was in 1:50 scale, so I scaled it with the scale command and conation option because otherwise, it does not allow me to do it directly.

3- I move it and position it between the axes

4- I start drawing rectangles using the reference points in the DXF

5- any sudden movement or action Freecad closes.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-12:
I don't know how to reproduce the crash. Could you record a video?

yorikvanhavre, 2024-09-13:
[image]
No problem for me either...

Roy-043, 2024-09-13:
The red text in the image points to a bug though.

kpemartin, 2024-09-15:
Which DXF importer, legacy or C++?
The attached sample file is already a FreeCAD save file, it seems to me that to reproduce this the original imported DXF file would be required.
I'm also wondering about about the red error text about being in `exportIFC.py`... how did we get into IFC-land?

gianlucafiore, 2024-09-15:
[image]

I have realized that the underlying problem is having the intersection snaps and line extensions or references activated, and when passing the mouse with any draft tool running over the imported DXF plane, the work becomes extremely slow, unstable and at any additional minimal movement, the program collapses and closes.

Roy-043, 2024-11-26:
The DXF that shows the issue is the result of importing a PDF. Every dash of every line is a separate object. Even texts are exploded line segments. Importing this DXF in FreeCAD results in a document that is hardly usable. A user can expect this, and probably more issues, when importing such a bad DXF. I am closing this.

closed 2024-11-26 by Roy-043

kpemartin, 2024-11-26:
I beg to disagree with closing this, at least for the reasons given. There is apparently nothing "bad" about the DXF, it just contains many objects, and the more appropriate response would be to treat this as a performance issue.

Perhaps there should be a closing code for Issues called something like "performance limitation" for this sort of thing.

#16736 alludes to the cost of parametric objects, and although FreeCAD supports parametric modelling, and such models are generally understood to suffer greatly when there are many objects, FreeCAD also has non-parametric objects which it should be able to handle in large numbers, and these are the sort of objects that are created by the DXF import process (without grouping).

That being said, as far as I can tell, this particular issue does not contain enough information to reproduce this problem because it appears that the original DXF file is not available, and the .FCStd file which is attached to the Issue loads and works fine, albeit being a pretty clunky drawing because of the way it was generated. So it could also be closed as "Unable to reproduce" if that is a choice.

luzpaz, 2025-12-15:
> That being said, as far as I can tell, this particular issue does not contain enough information to reproduce this problem because it appears that the original DXF file is not available, and the [...]

@gianlucafiore can you provide the original DXF file ?

gianlucafiore, 2025-12-15:
It's been a while since this happened, I don't know if I have it, I have limited availability to evaluate it, I'll reply as soon as I can