FreeCAD/FreeCAD #16487: Moving panel with mouse starts from 0x0 and not from the current mouse position

pinkavaj, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is a problem with placement of panels (eg. Report View, but applies for all panels). They can be dragged by mouse to various positions and they either snap to edge or they can be placed "floating" in the middle of the screen. When they are placed in the middle of the screen (floating), it is difficult to drag the window and place it to proper position, because the blue rectangle appears at top left corner of the screen, but the mouse is in the center of the screen. This disallows to move blue rectangle far enough to right or bottom edge, because the mouse hits the edge much sooner than the blue rectangle.

* OS: Arch Linux
* Graphic Platform: Wayland

### Full version info

```shell
OS: Arch Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38558 (Git)
Build type: Release
Branch: main
Hash: aef8e232af1f38c7ae8adc95756a35c0828177ba
Python 3.12.5, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
Screencast would be super helpful

pinkavaj, 2024-09-13:
https://github.com/user-attachments/assets/c2aacac4-6a7e-4b18-b533-ce9f80e04d7d

pinkavaj, 2024-09-13:
Notice I have seen, in recent few montsh/years, similar errors in other programs (games, mostly) and they were related to the fact Wayland is not leaking mouse/window position (so Qt cannot get the position and returns 0x0). Might this be that case?

luzpaz, 2024-09-13:
@kadet1090 have you seen this before ?

pinkavaj, 2024-09-18:
After some investigation I'm now pretty sure it is related to wayland. The QDockWidget, when set as floated is "isolated" window from Wayland point of view. The events `mousePressEvent/mouseMoveEvent` gets both relative and global position identical. AFAIK there is no (reasonable) way to get the global pointer position on Wayland and `mapToGlobal` is useless for this purpose.

pinkavaj, 2025-08-05:
This seems to be fixed now.

closed 2025-08-05 by pinkavaj