FreeCAD/FreeCAD #16488: PartDesign Helix: Thread start/end options.

leoheck, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The PartDesign Helix could have options to modify the Start/End of the thread. 

I want to have a smooth transition but my attempts using pad, sweep+helix and other methods are failing to produce a working feature. 

Sometimes they work. Sometimes they don't. So I was thinking that, maybe this could be an option to the PartDesign Helix itself.

For instance, having a checkbox to smooth the Beginning of the thread, another for the end would do the trick. Another to make them with the same settings or having different settings would make sense.

The length of this smoothness could be a parameter.

Another parameter could be smoothing adding mass increasing/decreasing the overall thread length or removing mass keeping the start/end at the same place.

### Full version info

```shell
1.0RC1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-13:
@FlachyJoe FYI

FlachyJoe, 2024-09-13:
Shown but I can't see why we'll add a such high level capability. 
More, if it doesn't work with existent simple operation it has no chance to works better with an integrated complicated process.

leoheck, 2024-09-13:
> Seen but I can't see why we'll add a such high level capability.

Because it is really hard for the end user to make this.
Also, since the feature makes the helix it is using, it already knows how to grow or shrink the ends of the thread.

Here is a video with a guy modeling this on Fusion (check minute 4)
https://forums.autodesk.com/t5/fusion-design-validate-document/form-a-guide-on-end-of-thread/td-p/9126627
I did not try to replicate this way yet.

FlachyJoe, 2024-09-13:
You can make another helix with the end face of the thread and a shifted axis
[image]
[Thread-end remove .zip .FCStd.zip](https://github.com/user-attachments/files/16995580/Thread-end.remove.zip.FCStd.zip)

leoheck, 2024-09-13:
hm that's pretty nice, indeed. I have to test this on my design, thanks. now, since the solution is straight forward, it would be a good implementation to push this feature to the next level.

leoheck, 2024-09-14:
@FlachyJoe, thanks. I was able to use this method on both ends of the thread. 
Tried many things until I could make this, but none of them gave erros or broke Freecad.
[image]

The result is a bit slow to be generated even on my computer that is quite fast, or maybe this version of Freecad is lagging a bit harder than usual.

leoheck, 2024-09-14:
@FlachyJoe your method could be added in this wiki https://wiki.freecad.org/Thread_for_Screw_Tutorial

FlachyJoe, 2024-09-14:
> @FlachyJoe your method could be added to this wiki https://wiki.freecad.org/Thread_for_Screw_Tutorial

Yes, feel free to do this.

luzpaz, 2024-09-14:
@leoheck if you write out some content I'll add it to the wiki

Baffardello2024, 2024-12-09:
G
[image]
[image]
[image]
Good morning thanks for the library, but in the new freecad 1.0 version it doesn't work perfectly, especially the printing doesn't work. do you have time to check? thanks in advance Luigi
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * AdditionalPythonPackages
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.12

luzpaz, 2024-12-11:
@Baffardello2024 please open a new ticket for this + zip up the .FCStd file and drag-drop it into github textbox