FreeCAD/FreeCAD #16489: overlay shows when opening a spreadsheet from treeview

jag45, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening a spreadsheet from the treeview the overlay shows on top of the spreadsheet until a cell is selected as per the screenshot. This also happens with the task panel if it is open at the time of opening the spreadsheet.
The "Auto hide in non 3D view" setting is ticked in preferences.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.65
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * dodo 1.0.1
  * frame 0.1.1
  * FreeCAD-themes 2024.7.24
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
Related to #15048 ?

Syres916, 2024-09-13:
I can't test for a few hours but I'm wondering if https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Spreadsheet/Gui/SpreadsheetView.cpp#L247-L248 should be a `return false` like in the Start module in this [commit](https://github.com/FreeCAD/FreeCAD/commit/bfdcc971580b5dd09913b7f9b923228728000298). The user can still get the Model/tree dockwidget open by swift mouse stroke and hold on the left extremity of the window.

jag45, 2024-09-13:
Hi Luzpaz
It's similar but not exactly the same. The overlay does not go away until you click on a cell in the spreadsheet. Also as in my screenshot the overlay is set to transparent but when you go to the spreadsheet it reverts to not being transparent.
Happens in "Freecad Dark" and "Freecad Light" themes, but in the Classic theme it seems to be semi-transparent as below.

[image]

Jon

luzpaz, 2024-09-15:
@kadet1090 can you weigh in?

kadet1090, 2024-09-15:
Yes, this is very similar issue. We can replicate the solution here also, but we should think about a better one.

My thought is to provide another Property type that is optional and can lazily fetch the value from preferences if user has not specified the color directly. The fetching of the default value could be defined as lambda in feature that is registering the property:

```
auto hGrp = App::GetApplication().GetParameterGroupByPath("User parameter:...")'
colorProperty->setDefaultValueProvider([hGrp]() {
    return unpackColorFromUserPreference(hGrp->GetUnsigned("..."));
});
```

Unless user provides value it would not be persisted in the document, user should also be able to clear the value if needed (reset to default).

@wwmayer what do you think about that proposed solution?

jag45, 2024-09-26:
Hi
There doesn't seem to be a lot happening on this. I might be misunderstanding the comment fromKadet1090 above, but the issue has nothing to do with colour.
It has to do with switching to a spreadsheet from the Treeview.

To replicate the issue:- 
1)  Unzip the attached Upright.FCStd file and load .
2)  click on "input sizes"
3)  click on "sizes"

This will bring up the spreadsheet. Notice that the spreadsheet is displayed with the front columns hidden behind the Treeview window.
Clicking any cell on the spreadsheet will hide the Treeview window and display the spreadsheet normally.
Hope this helps you find a solution.

[Upright.zip](https://github.com/user-attachments/files/17151430/Upright.zip)

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: MY-Darker-Flat.qss/MY-Darker-Flat/Fusion
Installed mods: 
  * A2plus 0.4.65
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * dodo 1.0.1
  * frame 0.1.1
  * FreeCAD-themes 2024.7.24

luzpaz, 2025-06-05:
Bumping for visibility.

closed 2025-08-20 by github-actions[bot]

JohnOCFII, 2026-01-22:
Seems to work properly for me now, on macOS.

```
OS: macOS 26.2
Architecture: arm64
Version: 1.2.0dev.45082 (Git)
Build date: 2026/01/20 20:22:29
Build type: Unknown
Branch: main
Hash: dee4ab6dbd844e76309ddde5a3def638d28dc789
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * Telemetry 1.0.4
  * Curves 0.6.70
```

FEA-eng, 2026-01-22:
On Windows (but I don't have to click on a cell - just move the mouse away from the panel):

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260120 (Git shallow)
Build date: 2026/01/20 21:54:07
Build type: Release
Branch: (HEAD detached at 9beca43)
Hash: 9beca4363ad327e453ee5a27f737537e31467901
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-09-22:
Still relevant, at least partially:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```