FreeCAD/FreeCAD #16491: Stylesheet: Classic theme black menu, broken Start page and weird input fields

mdrost, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I compiled FreeCAD on Windows 11 and encountered a bunch of bugs regarding Classic theme:

-----
1\) In menu "File" and "Edit" buttons are rendered as black rectangles.
[image]

-----
2\) In Start page if I jump from Classic theme to Light theme, then to Dark theme and back to Classic theme, then start page becomes broken i.e. all text and input fields became white/transparent.
[image]

This behavior does not happen if I jump from Classic theme to Light theme and back to Classic theme, or from Classic theme to Dark theme and back to Classic theme.
The page is restored to normal look if I jump to Dark theme, then to Light theme and back to Classic theme.

-----
3\) Different input fields during edit mode render old value in the background.
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38691 (Git)
Build type: Release
Branch: main
Hash: a4313f358ee56adee5c98bfc0a97c9fdba3cc7cf
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-15:
@MisterMakerNL have you seen this reported before ?

MisterMakerNL, 2024-09-15:
Could be because classic uses win11 qtstyle, and at them self recommend fusion style.
Because they even say win11 style I'd broken.
But it could also be something different.
Preference packs often need a restart, which probably would be a good idea here too, could you test this?

mdrost, 2024-09-15:
If I set Classic theme and restart FreeCAD then issues remain. However... if I set Light or Dark theme, restart FreeCAD and then set Classic theme then there are no issues. After restart issues come back.

luzpaz, 2024-09-15:
@MisterMakerNL an aside, should we just set Fusion style by default for Windows 11 as part of vanilla install ?

mdrost, 2024-09-15:
Is there a way I can print effective stylesheet for a given widget?

MisterMakerNL, 2024-09-16:
> If I set Classic theme and restart FreeCAD then issues remain. However... if I set Light or Dark theme, restart FreeCAD and then set Classic theme then there are no issues. After restart issues come back.

Seems to be that it indeed us a QStyle issue. Feels like Qtstyle is missing a bunch of settings that the other themes apply and they stay active untillyou restart.
There is currently no option to set qtstyle specifically for windows 11, which would probably be the preferred solution. Not sure if @maxwxyz has similar results on his win11. My PC doesn't support win11.

maxwxyz, 2024-09-16:
I've had the double text on input fields in the past

maxwxyz, 2024-09-16:
@Syres916 only Overlay

luzpaz, 2024-09-22:
>>> I've had the double text on input fields in the past

>> In both Overlay and non-Overlay or only in Overlay mode?

> Only Overlay

Should we open a separate ticket for this issue ?

Syres916, 2024-09-23:
Probably because the double text applies to both Qt5 and Qt6 whereas the black menu I believe only applies to Qt6 on Windows, my Qt6.7.2 isn't affected on EndeavourOS.

MisterMakerNL, 2025-06-05:
It's hidden in windows, because windows 11 Qstyle is buggy.

luzpaz, 2025-06-05:
Ok, so we can close this as a 'Wont Fix' because we aren't addressing non-Fusion Qstyles, right ?

closed 2025-08-20 by github-actions[bot]