FreeCAD/FreeCAD #16493: Subtractive helix adding material, generally not working (1.0RC1 regression?)

kbrantley, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Starting in 0.21.1, create a new drawing, part, and then a new sketch on the XZ plane. Create a simple shape, kind of like this:
[image]

Revolve the shape around the Z axis to create a solid object:
[image]

Set the transparency of the revolution to 90, and then create a new sketch on the XZ plane:
[image]

Using the new sketch, apply a subtractive helix, 10mm pitch, 33mm height, reversed:
[image]

To create internal threads on the cylinder:
[image]

Great. It worked. Load it up in 1.0RC1:
[image]

Recompute the helix, and observe how it just absolutely explodes, adding material where there shouldn't be any, and generally just doing things I don't understand:
[image]
[image]

Note: creating this from scratch in 1.0RC1 produces the same, buggy behavior.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

This is the file as generated by 0.21.1:
[HelixBug.zip](https://github.com/user-attachments/files/16985745/HelixBug.zip)

And this is the file after recomputing the object in 1.0RC1 and saving it:
[HelixBug-1.0RC1.zip](https://github.com/user-attachments/files/16985755/HelixBug-1.0RC1.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-13:
Suplicate of https://github.com/FreeCAD/FreeCAD/issues/14901

closed 2024-09-13 by maxwxyz

FlachyJoe, 2024-09-13:
@kbrantley It's a know bug of OCC for solid boolean operation with coincident elements. Tweak the sketch lengths by 1µm and it works:
[image]

PartDesign helix uses now a solid algorithm instead of the previous shell one. It fails on some cases but it's roughly better.