FreeCAD/FreeCAD #16494: Crash in Spreadsheet when all cells are selected and center is used

lucas-bueno777, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As the title indicates, bad performance starts to show when I create a new Spreadsheet and select all cells. If I use some command such as Center text, FreeCAD freezes and does not come back.

Steps to reproduce:

1. Create a new document
2. Insert a new Spreadsheet
3. Ctrl+A to select all cells (note that here the performance starts to drop)
4. Select the button to center text
5. FreeCAD freezes

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-12:
Related to #15673

xtemp09, 2024-09-13:
Strictly speaking, freeze does not take place. When you select all cells and apply something, _FreeCAD_ enumerates all the cells in the spreadsheet, taking some time to go through 7.4 GiB of memory (26 * 26 * 16384 * 700 bytes). It is carried out in the GUI thread; therefore, making UI unresponsive. _FreeCAD_ returns back to normal when it is finished.

luzpaz, 2024-09-13:
Thanks @xtemp09 that confirms that this is a dupe of #15673

closed 2024-09-13 by luzpaz