FreeCAD/FreeCAD #16495: CAM: Deburr operation on edges fails in some cases

jffmichi, 2024-09-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The deburr operation fails when used on edges in some specific cases. I could not figure out what exactly causes the failure. In the attached sample file [cam_deburr_issue.zip](https://github.com/user-attachments/files/16986044/cam_deburr_issue.zip) I'm trying to deburr a simple edge consisting of two ellipse arcs joined by a straight line. All lines are on z=0. Not sure what else to check...

The issue is quite annoying because in some cases it works and changing a minor detail on the model suddenly causes the failure.

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38402 (Git)
Build type: Release
Branch: main
Hash: 441fbdd064796fb412e35287a532d7fa98b6449e
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * kicadStepUpMod 11.2.7
  * dodo 1.0.1
  * sheetmetal 0.4.24
  * fcgear 1.0.0
  * fasteners 0.5.27
  * Assembly4 0.50.15
  * Curves 0.6.45
  * lattice2 1.0.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2025-06-24:
@pieterhijma I suspect this is an offsetting problem in OCCT.  Might want to add it to your dashboard.

tarman3, 2025-06-24:
Maybe latter
At this moment I am completely immersed in LeadInOut

sliptonic, 2025-06-24:
> Not sure where this would be best fixed. I guess ideally `makeOffset2D` would be able to handle all valid geometry. However, since those geometry functions seem to be a "little" tricky, I'll take a quick fix for a specific issue any day.
> 
> The idea that FreeCAD can only handle lines and arcs and you're not allowed to use ellipses or spines seems wild... it's valid geometry and FreeCAD fails to work with it, that's a bug that should be somehow/somewhen fixed.

You don't have to convince me.  FreeCAD in general works with ellipses and splines just fine.  It's just makeOffset2D that is the problem.  If it worked reliably, we wouldn't need libarea at all.  Alas, we're stuck with the geometry kernel that we have and OCCT is weak in this area.

That said, most gcode interpreters only use G1 and G2/3.   While linuxcnc, and I'm sure other commercial controls can technically do spline gcode (G5), it's almost never done.  The trajectory planners are good enough to take simpler gcode and process it without slowdown.  Since we want to support simpler controls like GRBL and Marlin, we need to reduce complex geometry to arcs and lines anyway

jffmichi, 2025-08-24:
.

tarman3, 2025-08-24:
[image]

tarman3, 2025-08-24:
Please check PR #23331

stefano2734, 2026-09-19:
I see here not the solved version after first issue with occt 7.8.1. 
is it next version 7.9.1 to 7.9.3  or new 8.0.0?

davidgilkaufman, 2026-09-21:
The fix for this ticket was #28645, no OCCT version changes required.