FreeCAD/FreeCAD #16498: SAT import error in 1.0.0RC1

g3grau, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,

already posted it on the forum, but meanwhile had a brief look into the issue:
Importing a sat model (to avoid posting the file, you can download it here: https://www.mirrorsemi.com/M-QFN48W.4.html) in the weekly build (appimage) failed with this error message:

Code: [Select all](https://forum.freecad.org/viewtopic.php?t=90512#)

10:04:40  Importing: /home/ggrau/Downloads/M-QFN48W.4.sat
10:04:40  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 113, in insert
    reader = read(filename)
             ^^^^^^^^^^^^^^
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerIL.py", line 46, in read
    if (importerSAT.readText(filename)):
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ggrau/.local/share/FreeCAD/Mod/InventorLoader/./importerSAT.py", line 160, in readText
    with open(fileName, 'rU') as file:
         ^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: invalid mode: 'rU'

Apparently the U can be just removed in Python3?
I stepped back to 0.21.2 and there the import works (with both of the offered approaches).

best regards
Guenter

### Full version info

```shell
Code: [Select all](https://forum.freecad.org/viewtopic.php?t=90512#)

OS: openSUSE Tumbleweed (KDE/plasma6)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Silk 1.0.0
  * sheetmetal 0.2.61
  * kicadStepUpMod 10.17.1
  * CfdOF 1.21.6
  * Assembly4 0.12.6
  * fasteners 0.4.54
  * offline-documentation 1.0.0-alpha
  * MeshRemodel 1.8919.0
  * LCInterlocking
  * parts_library
  * 3D_Printing_Tools
  * A2plus 0.4.60k
  * 3DfindIT 1.2.0
  * EM
  * ExplodedAssembly
  * ThreadProfile 1.84.0
  * workfeature
  * Lithophane
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Assembly3 0.11.4
  * InventorLoader 1.3.0
  * FEMbyGEN 1.2.0
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * frame 0.1.0
  * FEM_FrontISTR 0.1.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * pyrate
  * OpticsWorkbench 1.0.9
  * CommandPanel
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-13:
Can you propose a more elaborate solution ?

g3grau, 2024-09-13:
Hi,

sorry if I added strange flags..might have been my jackdaw running over the keyboard?!

According to [open()](https://docs.python.org/3.11/library/functions.html#open) there is no U argument for open(). Both unix and dos styles are handled anyway. So the line should be changed to `with open(fileName, 'r')` as of Python3.7 AFAIK. I guess the older version FC0.21 used 3.6 and 1.0 changed to 3.11? In that case the same issue may also occur elsewhere.

BR GG

Syres916, 2024-09-13:
Surely the most important thing is that this is not a core piece of code, it's an Addon and therefore the issue should have been logged [here](https://github.com/jmplonka/InventorLoader/issues) I'll be honest it doesn't look to be actively maintained, the last PR merged was in 2022.

luzpaz, 2024-09-14:
Closing as this is 3rd party related. 

CC @jmplonka

closed 2024-09-14 by luzpaz