FreeCAD/FreeCAD #16501: Draft: Annotation Styles Editor, Draft Preferences 'Scale multiplier' field in a confusing position

semhustej, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The 'Scale multiplier' field in Annotation Styles Editor is placed in the 'Units' group. But it does not influence units, it influences the whole annotation. In the properties panel it is placed under the 'Annotation' group. In global Draft Preferences the same variable is placed under the 'Texts' group. The same location is in the Task Panel, when 'Change default styles of objects' is invoked.
This is a bit confusing.

Annotation Styles Editor:
[image]

Draft preferences:
[image]

Task panel:
[image]

Element properties (the best placement in my opinion):
[image]

**Proposed solution:**
The 'Scale multiplier' influences not only text, and and it does not influence units. It influences all annotations, all values. Therefore, the location in the 'Annotation' group, as is in the element properties, makes the most sense. To achieve consistency and logical placement, I propose:

1. Place new group of properties in the **Annotation Styles Editor** with the name 'Annotation' at the top of the dialogue. This group would have only one value to set - 'Scale multiplier'. Remove 'Scale multiplier' from the 'Units' group
2.  Place new group of properties in the global **Draft Preferences -> Texts and dimensions** with the name 'Annotation' at the top of the dialogue. This group would have only one value to set - 'Scale multiplier'. Remove 'Scale multiplier' from the 'Texts' group
3.  Place new group of properties in the global  **'Change default styles of objects' ->  Task Panel** with the name 'Annotation' at the top of the dialogue. This group would have only one value to set - 'Scale multiplier'. Remove 'Scale multiplier' from the 'Texts' group

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-13:
Personally I agree with the proposed changes. Indeed having the scale factor inside "Units" is confusing.

hooperbloob, 2024-09-13:
Great write up and proposed solution.

furgo16, 2024-09-13:
Indeed, great analysis and proposal

PR #18299 (2024-12-05): Draft: change position of 'Scale multiplier' field in UI
Fixes #16501.
commit afdf0771 (2024-12-05): Draft: change position of 'Scale multiplier' field in UI

Fixes #16501.
closed 2024-12-05 by Roy-043