FreeCAD/FreeCAD #16503: Visibility of Link briefly toggles when a child changes visibility

NomAnor, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[visibility.webm](https://github.com/user-attachments/assets/f2de43c9-bdcd-494b-a692-0d4323bc4c73)

Changing the visibility of the Body (with _Space_ or the visibility icon) causes the visibility icon of the Link to briefly show that it is invisible.

This does not seem to cause any issues but might cause unnecessary updates/recomputes.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38690 (Git)
Build type: Release
Branch: integration
Hash: 6ac0cb75889b4909647392d41ec53a8740f43359
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.5.27
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-13:
@NomAnor can you upload an example file for reproduction ?

@pinkavaj have you seen this ?

NomAnor, 2024-09-13:
[VisToggle.FCStd.zip](https://github.com/user-attachments/files/16992451/VisToggle.FCStd.zip)

pinkavaj, 2024-09-13:
@luzpaz  I can reproduce the issue. It is reproducible not only for Body but also for any object like a Cube, Cylinder etc ...

It is reproducible in both ways, it does not matter if the visibility is changed in parnt or in link, the issue manifests. I did not observed any other sideeffect.

FEA-eng, 2026-10-05:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```