FreeCAD/FreeCAD #16505: Draft: Layer Manager/Layer property  Line width values

semhustej, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Draft Layer Manager, Line width values can be set as an integer between 1-99. 
When changing Line width in Layer properties, one can set float value between 0-99.99

When checking Line properties, Line Width values can be set as float between 1-10.
If a value of over 10 is set in Layer Manager, Line width defaults to 10. This is confusing behavior.

Relevant forum discussion: https://forum.freecad.org/viewtopic.php?t=90507

**Proposed change:**

1. Set value of Line width variable in Layer manager to float
2. If there is no other use to having  Line width < 10 and Line width > 1 (as is the case of line or polyline), limit the value range of Line width in Layer manager and Layer properties between 1 and 10. This would guarantee consistency and reduce user confusion.
3. If possible and applicable (I am not sure about this), ad 'px' as a unit to fields of Line width to Layer manager and Layer property. It seems the line width value is in pixels from my testing. This would limit user confusion, as some line widths are set directly in mm - such as those in techdraw workbench.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-13:
I don't remember when line width was changed from into to float in general UI, but 99% certain under the hood it's still an int expressing pixel values...

semhustej, 2024-09-13:
> I don't remember when line width was changed from into to float in general UI, but 99% certain under the hood it's still an int expressing pixel values...

Yes it seems like it from my testing. I was just eyeballing it, though. When changing line width in properties, it seems to stay the same when values like 3-3.5 are used. At 3.6 it jumps to a bit thicker line.

On this wiki page, Line Width is described as Float, this should be changed too:
https://wiki.freecad.org/Draft_Layer#View

yorikvanhavre, 2024-09-19:
> Yes it seems like it from my testing. I was just eyeballing it, though. When changing line width in properties, it seems to stay the same when values like 3-3.5 are used. At 3.6 it jumps to a bit thicker line.

Yes, that would be an indication that the value is actually integers (at 3.6 it gets rounded to 4). I think this needs a wider discussion at FreeCAD level and why the linewidth is expressed as float

closed 2025-08-06 by github-actions[bot]