FreeCAD/FreeCAD #16506: PartDesign: The color is reset

Pendelf, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The color is reset when editing the body further

https://github.com/user-attachments/assets/41c4a217-0b7b-4b11-b686-faa745ce3361

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.28
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-13:
This is another duplicate of https://github.com/FreeCAD/FreeCAD/issues/15503

maxwxyz, 2024-09-13:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/15503

closed 2024-09-13 by maxwxyz