FreeCAD/FreeCAD #16507: `Periodic B-spline by knots` crashes for 4 or more knots

iviar, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

"Periodic B-spline for 3 knots takes initiative to downgrade to three aligned(not those chosen) knots points at least editable but *crash* for more than 3 knots, with error:
/usr/include/c++/14.2.1/bits/stl_vector.h:1130: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38717 (Git)
Build type: Release
Branch: main
Hash: a4f742a2f72829eace09e7b7058a4615bc15b44b
Python 3.12.5, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * ExplodedAssembly
  * Curves 0.6.47
  * BOLTSFC
  * Silk 0.1.5
  * Estimate 0.1.2
  * Beltrami 1.1.0
  * Pyramids-and-Polyhedrons
  * fcgear 1.0.0
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Plot 2022.4.17
  * offline-documentation 1.0.0-alpha
  * Estimate.backup1717525195.739051 0.1.2 (Disabled)
```

### Subproject(s) affected?

Sketcher

### Anything else?

It's a while I'm seeing this problem in current (summer 2024) also fresh of today(2024-09-13) master git branch
any attempt to make a "Periodic B-spline by knots" shows the B-spline than it changes it and crash, or 3 points the construction is fine than something happen and it disappear, I'm possibly able to investigate it further up to a fix, but as I'm not familiar with the code I prefer to start rising the attention to this systemic crash

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-13:
Could you provide an example file and ideally a macro to record what you do? Does it work in the stable 0.21.2 version?

iviar, 2024-09-13:
About the macro file it doesn't finish to record because freecad crashes while recording the macro when the assert in stdlib is encountered,
i tried to modify a similar macro file (with 2 points to prevent the case of the crash) then tried to add a a third point in the macro file but the macro doesn't run,
so repro steps are (I thought was obvious from the description) and I will write them in the description as with the composing this comment I can't modify the description:
1. create a document
2. create a sketch
3. select for example XY plane (any would do)
4. left click "Periodic B-spline by knots" icon
5. left click on 4 or more random points in the sketch area (the more they are the longer it perform an intense computation before to crash)
6. right click to end the shetch (or press Esc)  [the last selected point get undone but that might be another issue]
wait a moment and it crash, all the time (with current git version)

archlinux switched recently to 1.0.0RC1.38642 detailed below (which the steps crash to me too, in doubt I launched also from another user that likely never used freecad and it crashed too (different machine also with archlinux))
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * ExplodedAssembly
  * Curves 0.6.47
  * BOLTSFC
  * Silk 0.1.5
  * Estimate 0.1.2
  * Beltrami 1.1.0
  * Pyramids-and-Polyhedrons
  * fcgear 1.0.0
  * fasteners 0.5.28
  * freecad.gears 1.2.0
  * Plot 2022.4.17
  * offline-documentation 1.0.0-alpha
  * Estimate.backup1717525195.739051 0.1.2 (Disabled)

Then I'm surprised to see that it crash also with (stable) 0.21.2! (I had to downgrade opencascade 1:7.8.1-1 -> 1:7.7.2-6 and fmt 11->10.2 to have coherent library to go back to test 0.21.2)
I've the impression that the crash might get triggered in the auto constraint phase

maxwxyz, 2024-09-14:
@AjinkyaDahale FYI

edwilliams16, 2024-09-14:
I was unable to reproduce this crash with

```OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.10.8
  * DynamicData 2.62.0
  * OpenTheme 2024.8.17
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.4.20
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.45
```

iviar, 2024-09-16:
gcc package owns the failing assert of /usr/include/c++/14.2.1/bits/stl_vector.h
as from https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/commits/main
ArcLinux gcc package uses snapshots of gcc 14.2.1 pre releases.
Maybe the scope of the crash is restricted to this weird gcc version choice of archlinux.

who don't experience the crash please provide a Macro of the repro steps (I couldn't produce a macro because of the crash, that I'm still experiencing nearly with any freecad version, with this gcc for the moment)

edwilliams16, 2024-09-16:
```
doc=App.ActiveDocument
doc.addObject('Sketcher::SketchObject', 'Sketch')
doc.Sketch.Placement = App.Placement(App.Vector(0.000000, 0.000000, 0.000000), App.Rotation(0.000000, 0.000000, 0.000000, 1.000000))
doc.Sketch.MapMode = "Deactivated"
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-44.553844,22.012373,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-20.045523,34.417820,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(17.775951,19.440510,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-9.304221,-22.314404,0)),True)
doc.getObject('Sketch').addGeometry(Part.Point(App.Vector(-38.804981,-11.119246,0)),True)
_finalbsp_poles = []
_finalbsp_knots = []
_finalbsp_mults = []
_bsps = []
_bsps.append(Part.BSplineCurve())
_bsps[-1].interpolate([App.Vector(-44.5538,22.0124),App.Vector(-20.0455,34.4178),App.Vector(17.776,19.4405),App.Vector(-9.30422,-22.3144),App.Vector(-38.805,-11.1192)], PeriodicFlag=True)
_bsps[-1].increaseDegree(3)
_finalbsp_poles.extend(_bsps[-1].getPoles())
_finalbsp_knots.extend(_bsps[-1].getKnots())
_finalbsp_mults.extend(_bsps[-1].getMultiplicities())
doc.getObject('Sketch').addGeometry(Part.BSplineCurve(_finalbsp_poles,_finalbsp_mults,_finalbsp_knots,True,3,None,False),False)
del(_bsps)
del(_finalbsp_poles)
del(_finalbsp_knots)
del(_finalbsp_mults)
conList = []
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',0,1,5,0))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',1,1,5,1))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',2,1,5,2))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',3,1,5,3))
conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',4,1,5,4))
doc.getObject('Sketch').addConstraint(conList)
doc.getObject('Sketch').exposeInternalGeometry(5)
constraintList = []
doc.recompute()
```

Makes a five point BSpline by knots successfully on my system.

AjinkyaDahale, 2024-09-17:
Cannot reproduce with
```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) Snap 1138
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: Dark-modern.qss/Dark modern/Qt default
```

Anyone has OCC 7.8.1?

AjinkyaDahale, 2024-09-17:
Cannot reproduce with:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 +44 (Git)
Build type: Release
Branch: sk-reduce-cognitive-complexity
Hash: f7008529121288892d687e3d1fe6cd2a1f0d6268
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.8.2.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Curves 0.6.47
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

iviar, 2024-09-17:
As in my previous comment the problem seems very specific to ArcLinux and its weird choice to use a snapshot version of gcc, afrer that comment nearly there was and there is not need to test on other linux distributions than archlinux if you want confirmation

> gcc package owns the failing assert of /usr/include/c++/14.2.1/bits/stl_vector.h as from https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/commits/main ArcLinux gcc package uses [...]

AjinkyaDahale, 2024-09-20:
> As in my previous comment the problem seems very specific to ArcLinux and its weird choice to use a snapshot version of gcc, afrer that comment nearly there was and there is not need to test on other linux distributions than archlinux if you want confirmation

OK. Noted. So can you confirm whether the macro provided by @edwilliams16 also crashes for you?

I'm myself on Manjaro. It has its own thing going w.r.t. packages but pretty close to Arch. So it's probably a bleeding edge issue.

iviar, 2024-09-20:
Yes the macro crashes in arclinux when executed within FreeCAD UI,
I wrote and email to 2 maintainers archlinux gcc,
despite I opened the bug I don't think anymore FreeCAD should care for this issue because being archlinux the only distribution case to encounter a failing assert seems because such unstable gcc it's an inappropriate choice from that distribution

closed 2024-09-20 by iviar