FreeCAD/FreeCAD #16512: freecad rc1 crashes while hovering the mouse upon 3d parts

littlett77, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

freecad rc1.0.0 versions 1.0.0RC1.38643 (Git) AppImage and 1.0.0RC1.38642 (Git) AppImage crash after loading the file BIMExample.FCStd and hovering the cursor over the many parts composing the model that are each subsequently selection-highlighted in yellow. This unespected behaviour does not happen with versions 0.22.0dev.38506 (Git) AppImage and 0.21.2.33771 (Git) AppImage.

### Full version info

```shell
[code]
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.9
  * CfdOF 1.27.4
  * 3D_Printing_Tools
  * fasteners 0.5.25
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Manipulator 1.5.7
  * OpticsWorkbench 1.0.19
  * LCInterlocking
  * BIM021 2021.12.0
  * btl 0.9.9
  * freecad.gears 1.2.0
  * 3DfindIT 1.2.0
  * Curves 0.6.42
  * ExplodedAssembly
  * sheetmetal 0.4.23
  * AirPlaneDesign 0.4.1
  * POV-Ray-Rendering
[/code]
...................................
ok
different POV

================================

OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.9
  * CfdOF 1.27.4
  * 3D_Printing_Tools
  * fasteners 0.5.25
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Manipulator 1.5.7
  * OpticsWorkbench 1.0.19
  * LCInterlocking
  * BIM021 2021.12.0
  * btl 0.9.9
  * freecad.gears 1.2.0
  * 3DfindIT 1.2.0
  * Curves 0.6.42
  * ExplodedAssembly
[... 147 lines cut ...]
#33  /tmp/.mount_FreeCAhQhuni/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7fba7ad9fbea]
#34  /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fba7eb10ad7]
#35  /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fba7eb13ac7]
#36  /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fba7eb140ce]
#37  0x7fba8030aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/libQt5Core.so.5+0x66
#38  0x7fba802b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/libQt5Core.so.5+0x11b
#39  0x7fba802bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7fba82e9f943 in Gui::Application::runApplication() from /tmp/.mount_FreeCAhQhuni/usr/bin/../lib/libFreeCADGui.so+0x983
#41  /tmp/.mount_FreeCAhQhuni/usr/bin/freecad(+0x7640) [0x555aec87f640]
#42  /usr/lib/libc.so.6(+0x25e08) [0x7fba7fe34e08]
#43  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7fba7fe34ecc]
#44  /tmp/.mount_FreeCAhQhuni/usr/bin/freecad(+0x79a9) [0x555aec87f9a9]
```

### Subproject(s) affected?

None

### Anything else?

Freecad version 0.21.2 by loading that example file puts the 3d camera on a different point of view than the other tested versions.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

littlett77, 2024-09-13:
nice sw :)

xtemp09, 2024-09-13:
The backtrace seems to be connected to #16232.

luzpaz, 2024-09-13:
@littlett77 can you test the weekly ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

littlett77, 2024-09-14:
Just tried version 1.1.0dev.38728 (Git) AppImage and the issue is not an issue anymore.
Also tried against the test file provided here: https://github.com/FreeCAD/FreeCAD/issues/16232#issuecomment-2325802506

Thanks for the fix.

OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.9
  * CfdOF 1.27.4
  * 3D_Printing_Tools
  * fasteners 0.5.25
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Manipulator 1.5.7
  * OpticsWorkbench 1.0.19
  * LCInterlocking
  * BIM021 2021.12.0
  * btl 0.9.9
  * freecad.gears 1.2.0
  * 3DfindIT 1.2.0
  * Curves 0.6.42
  * ExplodedAssembly
  * sheetmetal 0.4.23
  * AirPlaneDesign 0.4.1
  * POV-Ray-Rendering

littlett77, 2024-09-14:
Thanks for the fix.

closed 2024-09-14 by littlett77

luzpaz, 2024-09-14:
> nice sw :)

:monocle_face: