FreeCAD/FreeCAD #16516: FEM: "Use FreeCAD material editor" not working

FEA-eng, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed a serious issue (IMO RC blocker if confirmed) in the newest weekly build:
1. Go to FEM. Add Analysis container.
2. Add "Material for solid" or "Material for fluid".
3. Click "Use FreeCAD material editor". Errors will appear:

```
19:54:27  Traceback (most recent call last):
19:54:27    File "E:\FreeCAD_weekly-builds-38684-conda-Windows-x86_64-py311\Mod\Fem\femtaskpanels\task_material_common.py", line 331, in edit_material
19:54:27      new_material_params = MaterialEditor.editMaterial(
19:54:27                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19:54:27    File "E:\FreeCAD_weekly-builds-38684-conda-Windows-x86_64-py311\Mod\Material\MaterialEditor.py", line 966, in editMaterial
19:54:27      editor = MaterialEditor(material=material, card_path=card_path, category=category)
19:54:27               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19:54:27    File "E:\FreeCAD_weekly-builds-38684-conda-Windows-x86_64-py311\Mod\Material\MaterialEditor.py", line 72, in __init__
19:54:27      self.widget = FreeCADGui.PySideUic.loadUi(filePath + "Resources" + os.sep + "ui" + os.sep + "materials-editor.ui")
19:54:27                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19:54:27    File "<string>", line 4, in <module>
19:54:27  RuntimeError: Cannot open file
```

@marioalexis84 and @davesrocketshop FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-13:
Confirmed using:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git) AppImage
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * GDML 2.0.0
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

Syres916, 2024-09-13:
This is the [PR](https://github.com/FreeCAD/FreeCAD/pull/16322) that's WIP to fix the issue to the best of my knowledge.

johnny555, 2024-09-17:
Hi, I got a similar error in FreeCAD 1.0.0. RC 1 on Windows. 

10:03:51  Running the Python command 'CalculateMassAndInertia' failed:
Traceback (most recent call last):
  File "C:\Users\johnv\AppData\Roaming\FreeCAD\Mod\freecad.cross\freecad\cross\ui\command_calculate_mass_and_inertia.py", line 48, in Activated
    default_material = material_from_material_editor(robot.MaterialCardPath)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\johnv\AppData\Roaming\FreeCAD\Mod\freecad.cross\freecad\cross\freecad_utils.py", line 738, in material_from_material_editor
    material_editor = MaterialEditor.MaterialEditor(card_path=card_path)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\johnv\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Material\MaterialEditor.py", line 72, in __init__
    self.widget = FreeCADGui.PySideUic.loadUi(filePath + "Resources" + os.sep + "ui" + os.sep + "materials-editor.ui")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 4, in <module>

Using the Cross - Ros workbench to add material types.

drfenixion, 2024-09-21:
Confirm in 1.1.0dev.

And looks like there is fix for that [https://github.com/FreeCAD/FreeCAD/pull/16322](https://github.com/FreeCAD/FreeCAD/pull/16322)

FEA-eng, 2024-09-22:
@maxwxyz or @luzpaz IMO this should be a Blocker.

FEA-eng, 2024-09-25:
Fixed by #16322.

closed 2024-09-25 by FEA-eng