FreeCAD/FreeCAD #16517: Part: Simple sketches don't show up as available profiles for lofting

theosib, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a really simple design here with three sketches that I want to loft. But they don't show up as available profiles when I select loft. Did lofting just completely break? I'll see about getting the next snapshot and see.

[failed loft.FCStd.zip](https://github.com/user-attachments/files/16997860/failed.loft.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-09-13:
I tried a bunch of different weekly builds. All of the 0.22 versions only list useless things like Std_Part axes and planes, which seems like its own bug. 0.21.2 lists nothing at all.

theosib, 2024-09-13:
Also, I did a geometry check on each of the sketches, and there are no problems. Also, they all extrude just fine. I just can't use them for lofting.

Roy-043, 2024-09-13:
The sketches contain a closed contour and a (normal geometry) point. This makes them invalid for a Part Sweep operation. They can only contain either. You need to remove the points or turn them into construction geometry.

Closing this as a user error.

closed 2024-09-13 by Roy-043

theosib, 2024-09-13:
Oh. I thought points were construction geometry by default. I guess I didn't notice when that changed. Sorry about that. And thanks for the help.

theosib, 2024-09-13:
BTW, do you think it would make sense to warn the user about this somehow? There's no way I was going to figure this out on my own.