FreeCAD/FreeCAD #16519: Crashing on Windows 11

Iliya5digital, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is in running the program unpacked from the archive. After unpacking from the archive, the program crashes 3 times. I attach 3 crash reports and 3 error reports. The video shows how it happened.

### Full version info

```shell
it could not be opened because it does not open
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/96b2083a-96b7-4238-b663-4981a8ecba35

[crash and error reporting.zip](https://github.com/user-attachments/files/16998459/crash.and.error.reporting.zip)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-09-14:
In the video, version 0.21

xtemp09, 2024-09-14:
From [^1]:

> The Windows 10 error ‘EXTERNAL EXCEPTION E06D7363’ occurs when a program tries to copy or move a locked file. It may be locked by Windows, your antivirus software or malware scanner, the error isn’t picky.

[^1]: https://forums.tomshardware.com/faq/how-to-fix-windows-10-error-%E2%80%98external-exception-e06d7363%E2%80%99.2752246/

maxwxyz, 2024-09-14:
@Iliya5digital can you try with the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-09-23:
@Iliya5digital :point_up_2:

maxwxyz, 2024-10-07:
any news on this? Otherwise I'll close this.