FreeCAD/FreeCAD #16523: [Sketcher] Regression: on entering Edit Mode with TouchPad Nav Style, holding Shift key and the slightest mouse movement makes the object jump in X positive, Y negative

Syres916, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On starting FreeCAD, click on the _PartDesignExample_ file to load it, select _Face6_ and in _Sketcher_ click the _New Sketch_ icon and confirm _Plane Face_ and click _OK_. As a Touchpad Nav Style user, my default method for aligning the origin is to hold the left _Shift_ key and move the mouse usually a small amount to align to the centre of the window. 

Having completed a Git Bisect, this [commit](https://github.com/FreeCAD/FreeCAD/commit/382ca30d1973658850f53720778bbe4064c8d4bc) which seems quite bizarre is the cause of the significant jump of the origin to the most lower right corner of the window with the most minimal mouse movement. The latest AppImage doesn't have this behaviour but of course is packaged very differently to my self compiled build.

It seems to me that there doesn't appear to be any reason why the change couldn't have a `# ifdef FC_OS_MACOSX` condition applied to it, @shaise your input would be appreciated.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * GDML 2.0.0
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-14:
I'm going to carry out a second Git Bisect, see https://github.com/FreeCAD/FreeCAD/issues/16223#issuecomment-2350069333 as there seems to be some similarities to the strange behaviour.

maxwxyz, 2024-09-14:
@Rexbas FYI

Syres916, 2024-09-14:
It's certainly an issue around camera focal distance etc, I can stop the issue by removing the `    viewer->setCameraOrientation(rot);` from https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp#L3466-L3483 obviously I'm then left will working at whatever camera angle was before going into Edit Mode. If I put the code back as is in main then even using the _View Fit All_ shortcut doesn't stop the misbehaviour. 

From a debugging point of view I get a _focal distance_ output of `87.0` and a _dist_ of `102.969948`

In 0.21 branch the same variables (working as expected) are:
_focal distance_ output of `94.684029` and a _dist_ of `59.686089`

@Rexbas I hope this feedback is of use.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Release
Branch: main
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33839 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: 1380323d8562578487d03091d3e2f572ac1d20ef
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```

Rexbas, 2024-09-15:
I cannot reproduce it

```
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 (Git)
Build type: Release
Branch: main
Hash: b4923d1c584a4dce794d91a4b090a5293569223a
Python 3.11.0, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.21
```

Syres916, 2024-09-15:
> I cannot reproduce it

Sorry to read that, I'll have a try at compiling Coin 4.0.3 tomorrow and if successful then compile FC against it to see whether there's any difference but I'm certainly not expecting any. Otherwise I'll be backporting lots of commits to my 0.21.x branch, it can't be a dependency because both are built on exactly the same libraries of the same system.

Rexbas, 2024-09-16:
I can reproduce it now. It only happens when the mouse is in the 3D view and has not yet moved after entering edit mode. If I enter edit mode, then move my mouse, and then press shift, it doesn't happen. Also, it looks like the object is moving more when the mouse is closer to the center before entering editing mode.

luzpaz, 2024-09-25:
Any fix in sight ?

Rexbas, 2024-09-25:
I am working on it.

Rexbas, 2024-10-06:
This regression was introduced in 22a98d81. I made a fix that waits for the second mouse movement after panning started so big jumps are prevented.

PR #17057 (2024-10-07): Gui: Fix Touchpad navigation style panning in edit mode
Fixes #16523.
closed 2024-10-07 by chennes