FreeCAD/FreeCAD #16525: DAG View not rendered properly

chrsoo, 2024-09-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following the instructions on [Activating the DAG View](https://wiki.freecad.org/DAG_view), the DAG View ends up all jumbled when displayed in the UI.

### Full version info

```shell
[code]
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * DynamicData 2.62.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

Screenshot of the jumbled DAG View:
[image]

This is the error I hope to solve by using the DAG View:
```
rpi_flat_white#Sketch022, rpi_flat_white#Pocket009, rpi_flat_white#Sketch023, rpi_flat_white#Pad010, rpi_flat_white#Mirrored001, rpi_flat_white#Pocket004, 
rpi_flat_white#Sketch018, rpi_flat_white#Pocket005, 

07:27:51  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: The graph must be a DAG.
07:27:56  367.954 <App> Document.cpp(2448): The graph must be a DAG.
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Mirrored001 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Pad010 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Sketch023 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Pocket009 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Sketch022 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Pocket005 still touched after recompute
07:27:56  370.077 <App> Document.cpp(2803): rpi_flat_white#Sketch018 still touched after recompute
07:33:51  722.774 <App> Document.cpp(2448): The graph must be a DAG.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-14:
@chrsoo could you test using the latet dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-09-14:
> but all the weekly build images fail verification by MacOS:

Workaround: https://github.com/FreeCAD/FreeCAD-Bundle/issues/291#issuecomment-2306592148

chrsoo, 2024-09-14:
@luzpaz - thanks, that works! @maxwxyz - I can confirm that the issue is still present in `1.1.0dev.38728`.

A side question, what's going on with the versioning? Current release is 0.21 and the weekly is 1.1. What happened to 1.0?

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Light-blue.qss/unset/Qt default
Installed mods: 
  * DynamicData 2.62.0
  * Help 1.0.3
```

luzpaz, 2024-09-14:
> Version: 1.1.0dev.38728 (Git)

Not sure. @adrianinsaval any idea ?

adrianinsaval, 2024-09-14:
1.0 will be released from a branch https://github.com/FreeCAD/FreeCAD/tree/releases/FreeCAD-1-0, so main will be the base for 1.1, fixes are still getting backported to the 1.0 branch

adrianinsaval, 2024-09-14:
I can't reproduce this issue, can you test with a clean config? does it still happen if you enable a theme? does it happen with any file or just specifically with the one you are having a problem with?

chrsoo, 2024-09-14:
@adrianinsaval - Resetting the preferences removes the DAG View.  Adding it back again following the [DAG View](https://wiki.freecad.org/DAG_view) documentation actually makes it work as expected, both for a new project and my problematic project with the cyclic graph that needs to be fixed.

Also tested both the 0.21 release version and the 1.1 weekly build - both works.

Not sure if the problem was caused by a configuration error caused by me doing something wrong when enabling the DAG View, or if there was some strange configuration combination that caused the problem. I am just starting out FreeCAD and have not made many configuration changes. If I manage to reproduce the issue, I will report it back here.

20/20 hindsight I should have have somehow saved my broken config before resetting... 😕

(BTW impressed by the very fast response - thanks for the help so far - FreeCAD is an impressive open source project!)

JohnOCFII, 2024-09-14:
> Edit: @JohnOCFII, could you confirm the dag issue on macOS ? TIA!

Still need me to try?  Or does the OP's response negate that need?

luzpaz, 2024-09-14:
@JohnOCFII no need. Thanks

adrianinsaval, 2024-09-14:
I'll close the issue for now since tehre's no reasonable action to take on our part but feel free to reopen if you manage to reproduce, in hindsight I should have told you to keep the config to try and figure out what was different! but without it it will be hard to know what combination can cause this.

macdroid53, 2025-07-01:
This seems to be back. this is what I get in Debian and Manjaro FC build 42464.

[image]

AndrewBuck, 2025-07-04:
I can confirm I am also getting this bug as well.  I am running on Linux Mint, originally I was running version 1.1.0.40267 installed via flatpak on my system and I just installed 'freecad-daily' via the PPA to test that version as well.  The bug appears in both versions.  It works correctly on a dog bed I modeled where the graph is a proper DAG, but my kitchen counter design has a cycle in the graph and that causes this bug (both versions seem to exhibit identical behavior).  Loading both files and clicking in the tree view on one or the other jumps between the correct DAG View for the dog bed, but the jumbled one when you select any part of the Kitchen Counter.

Looking closely at the screenshot posted by OP as well as my own posted below it looks like all of the items are just being rendered on top of each other.  I think what is happening is since there is a cycle there is no longer a defined top to bottom ordering and the tool just gives up and renders everything on the same line in a big jumbled mess.  The blob on the left is all the icons and the blur on the right shows the names of the objects from my model.

[image]

I think the issue should be reopened since it is present in multiple builds and now confirmed by 2 different users.

maxwxyz, 2025-07-04:
@hyarion can you reproduce on macOS?

FEA-eng, 2026-01-11:
I can't reproduce it on Windows, but it's likely OS-specific

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260106 (Git shallow)
Build date: 2026/01/06 15:36:19
Build type: Release
Branch: (HEAD detached at 9b64da8)
Hash: 9b64da827a112d88a025be26316e3d023ff491dc
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-09-09:
@aw-engineer can you please add your full [About](https://wiki.freecad.org/About) info?

aw-engineer, 2026-09-23:
> [@aw-engineer](https://github.com/aw-engineer) can you please add your full [About](https://wiki.freecad.org/About) info?

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 26.3.0dev.48073 (Git)
Build date: 2026/08/14 14:49:17
Build type: Release
Branch: additivepipe-binder
Hash: 23770f89ca2cd2ec7dc7651a2ede0342b49ab0bd
Python 3.13.5, Qt 6.8.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.4, Clipper2 , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: wayland/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/165.974/1
OpenGL version/vendor/renderer: 4.5 (Compatibility Profile) Mesa 25.0.7-2/Mesa/llvmpipe (LLVM 19.1.7, 256 bits)
```

luzpaz, 2026-10-03:
> Edit: after removing an extra Local Coordinate System in the drawing's single Body (which was problematic in an unrelated experiment) the DAGView is displaying nicely.

What would the diagnosis be in that case ?

aw-engineer, 2026-10-03:
> What would the diagnosis be in that case ?

There is no clear diagnosis, just an anecdotal report.