FreeCAD/FreeCAD #16526: Usability bug: Absolutely no way to discover from within FreeCAD that non-construction points and closed curves are incompatible for lofting

theosib, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please reference this closed bug report: https://github.com/FreeCAD/FreeCAD/issues/16517

What I'm reporting here is that this is a serious usability bug.

There is no means within FreeCAD itself by which one could discover that having both non-construction points and closed curves in the same sketch would silently make them unlisted for sweeps and lofts.

Instead, it is necessary to post in forums and be lucky enough to find someone who has this tribal knowledge. Otherwise, this is entirely unsolvable. Or at least I would not have figured this out myself.

My suggestion would be to allow these sketches to be included in lofts and sweeps but then throw an error when the user attempts to use one. This would allow the user to find out why the sketch can't be used and then fix the problem.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-09-14:
This really needs to be seen as a more general error-handling problem.

There is some code that is very specifically checking for this rule violation so as to exclude such sketches as available profiles. What reason is there that this same code can't fail with an error message that corresponds to the very check it's making? Any time you have a rule check, the failure should come with a description of that rule.

I think these sketches should be listed for lofts and sweeps, and then the rule check should be applied after the user tries to use one that violates the rule, providing a popup with the error at that time. But this error reporting should be part of a more general system that associates human-readable descriptions with all rule checks. No rule check should ever fail silently or without description that corresponds to the rule applied.

Another option, BTW, would be to add a warning to the geometry check. The problem is that most beginners would not have any awareness of the geometry check tool. So perhaps both of these suggestions could be applied.

Roy-043, 2024-09-15:
As explained in the forum there is no 'tribal knowledge' here. The behavior is documented.

IMO this is not well thought out. I don't want to be presented with a list where several of the option cannot be used. That would be a bad idea in terms of UI/UX.

theosib, 2024-09-16:
Would you say it's a bug to include Std_Part axes and planes?
[image]

Roy-043, 2024-09-16:
> Would you say it's a bug to include Std_Part axes and planes?

You have created a separate issue for this: #16572. So let's not discuss that here.