FreeCAD/FreeCAD #16528: FreeCAD RC 1.0 unable to pickup new python dependancies

KeithSloan, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For a development branch I needed a new python library pyg4ometry.

I added to FreeCAD-addons after the feature freeze and all is good with FC0.22 but not RC1.0
Which is fine given the feature freeze but what is the situation going forward? 

### Full version info

```shell
When I load the development branch fluka in RC1.0 it does not give access to pyg4ometry.

The addon manaager is not seeing any update, or pick up info on the new dependancy.

If I uninstall and reinstall the workbench it offers me the option of installing the new dependency but does not seem to find it.

Python 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 19:00:53) [Clang 16.0.6 ] on darwin
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_Workbench',9)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> # Gui.runCommand('Std_AddonMgr',0)
>>> import pyg4ometry
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD 1.0 RC.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'pyg4ometry'
>>> 

In the startup report view there is this warning

06:11:02  Attempting to fetch remote copy of ALLOWED_PYTHON_PACKAGES.txt...
06:11:02  Overriding local ALLOWED_PYTHON_PACKAGES.txt with newer remote version
06:12:23  QObject::moveToThread: Current thread (0x600003b6c670) is not the object's thread (0x0).
Cannot move to target thread (0x600003f78ec0)

06:12:23  You might be loading two sets of Qt binaries into the same process. Check that all plugins are compiled against the right Qt binaries. Export DYLD_PRINT_LIBRARIES=1 and check that only one set of binaries are being loaded.

I also note that some Macro's were removed
06:10:54  Removed extra installed file /Users/keithsloan/Library/Application Support/FreeCAD/Macro/ProcessDirStruct_Brep.FCMacro
06:10:54  Removed extra installed file /Users/keithsloan/Library/Application Support/FreeCAD/Macro/ExpandArrayCN3.FCMacro
06:10:54  Removed extra installed file /Users/keithsloan/Library/Application Support/FreeCAD/Macro/ExpandArray.FCMacro
06:10:54  Removed extra installed file /Users/keithsloan/Library/Application Support/FreeCAD/Macro/ExpandArray-MMH-5.FCMacro

I also note all the workbench license checks.
Maybe there should be a heads up to workbench developers as to what is now required?
06:10:19  Checking network connection...
06:10:20  GitHub's zen message response: Design for failure.
06:10:20  Attempting to normalize non-compliant license 'GPL-2'... found valid SPDX license ID GPL-2.0
06:10:20  Attempting to normalize non-compliant license 'GPL-2'... found valid SPDX license ID GPL-2.0
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:20  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'GPLv3'... found valid SPDX license ID GPL-3.0
06:10:20  Attempting to normalize non-compliant license 'LGPLv3'... found valid SPDX license ID LGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:20  Attempting to normalize non-compliant license 'lgpl-2.1-or-later'... found valid SPDX license ID LGPL-2.1-or-later
06:10:20  Attempting to normalize non-compliant license 'LGPLv3'... found valid SPDX license ID LGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'LGPL-3'... found valid SPDX license ID LGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'GPLv3.0'... found valid SPDX license ID GPL-3.0
06:10:20  Attempting to normalize non-compliant license 'GPLv2.1'... failed to normalize (typo in ID or invalid version number??)
06:10:20  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
06:10:20  Attempting to normalize non-compliant license 'AGPLv3.0'... found valid SPDX license ID AGPL-3.0
06:10:20  Attempting to normalize non-compliant license 'CCOv1'... failed to normalize (typo in ID or invalid version number??)
06:10:20  Attempting to normalize non-compliant license 'GPL-3'... found valid SPDX license ID GPL-3.0
06:10:20  Invalid version specified for tag version in Addon Trails: 2022-01
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:20  Attempting to normalize non-compliant license 'LGPLv2.1'... found valid SPDX license ID LGPL-2.1
06:10:21  Attempting to normalize non-compliant license 'GPL-2'... found valid SPDX license ID GPL-2.0
06:10:34  Saved slider position at 3528
06:10:37  pip timed out
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KeithSloan, 2024-09-14:
Not sure how it got there but just noticed a copy of AM_INSTALLATION_DIGEST.txt in the workbenches Repro
Don't see why FC would pick this up. Have now removed from the workbench repro

maxwxyz, 2024-09-14:
@KeithSloan does it work now?

KeithSloan, 2024-09-14:
Appears not I think the AM_INSTALLATION_DIGEST in my Repo is a red herring, like I say no idea how it got there.

When I checked with Addon Manager it does not notice any new dependency.
If I uninstall and try reinstall the workbench, it picks up on missing python libs but then barfs with this popup.
Which I note mentions 0.22 although I am running RC1.0

[image]

maxwxyz, 2024-09-14:
@chennes FYI

KeithSloan, 2024-09-14:
Okay as the install of the workbench failed in RC1.0.

I went back to FC 0.22 (Need to work on workbench) and it was uninstalled so used FC0.22 Addon Manager to Install and the  AM_INSTALLATION_DIGEST is back in my local copy of the branch

Last login: Sat Sep 14 08:01:25 on ttys002
cd ~/Library/"Application Support"/FreeCAD/Mod/GDML/freecad/gdml;git status     
keithsloan@Keiths-iMac-2 ~ % cd ~/Library/"Application Support"/FreeCAD/Mod/GDML/freecad/gdml;git status
On branch Main
Your branch is up to date with 'origin/Main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	../../AM_INSTALLATION_DIGEST.txt

nothing added to commit but untracked files present (use "git add" to track)
keithsloan@Keiths-iMac-2 gdml %

Syres916, 2024-09-14:
@KeithSloan surely your Package.xml has to have the dependency added such as from another of your own repositories https://github.com/KeithSloan/OpenSCAD_Alt_Import/blob/master/package.xml#L33

KeithSloan, 2024-09-14:
Yes and it was in commit https://github.com/KeithSloan/GDML/commit/797a21d25a747231bc5cb0ec162b6be3c1b5f886

Don't understand how that update is now missing

Syres916, 2024-09-14:
Maybe https://github.com/KeithSloan/GDML/pull/139 will assist

KeithSloan, 2024-09-14:
> Yes and it was in commit [KeithSloan/GDML@797a21d](https://github.com/KeithSloan/GDML/commit/797a21d25a747231bc5cb0ec162b6be3c1b5f886)
> 
> Don't understand how that update is now missing

Okay missed your comment was about package.xml not metadata.txt.

 I had added to metadata and that worked in FC0.22

Is the requirement now that python packages have to be added to metadata.txt  AND package.xml

note: OpenSCAD dependancies was sorted out by @onekk and he had numerous discussions with Chris @chennes

KeithSloan, 2024-09-14:
Okay workbench updated with your pull request, still fails to install on RC1.0

Not happy about pip and looking 0.22

If I say install anyway it just locks out

[image]

Syres916, 2024-09-14:
Well using the AppImage created earlier this morning, I know that the _pip_ command is executing correctly on my system because I installed _ezdxf_ as a dependency of _InventorLoader_ addon. When I look at the documentation for _pyg4ometry_ https://www.pp.rhul.ac.uk/bdsim/pyg4ometry/installation.html I'm seeing a lot of build dependencies which is slightly alarming. Until I can dual boot back into my Windows partition (at present a large Git Bisect is under way) I'm not going to be able to a definitive reason for this library behaving differently.

KeithSloan, 2024-09-14:
pygometry is only required in a development branch, so currently not used in Main branch, it is just there so that I can easily checkout the development branch

Syres916, 2024-09-15:
I rebooted into my Windows partition and ran my own pip install macro so I could see all the feedback during the install of _pyg4ometry_, even though my Python 3.8 FreeCAD 1.0rc1 build appears to start the checks fine, it stills errors:

```
07:10:52  Collecting pyg4ometry
  Downloading pyg4ometry-1.1.2.tar.gz (6.1 MB)
     ---------------------------------------- 6.1/6.1 MB 4.6 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Installing backend dependencies: started
  Installing backend dependencies: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting importlib-resources (from pyg4ometry)
  Using cached importlib_resources-6.4.5-py3-none-any.whl.metadata (4.0 kB)
Collecting antlr4-python3-runtime==4.9.3 (from pyg4ometry)
  Using cached antlr4_python3_runtime-4.9.3-py3-none-any.whl
Collecting configparser (from pyg4ometry)
  Downloading configparser-7.1.0-py3-none-any.whl.metadata (5.4 kB)
Requirement already satisfied: matplotlib in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from pyg4ometry) (3.3.3)
Collecting networkx>=3 (from pyg4ometry)
  Downloading networkx-3.1-py3-none-any.whl.metadata (5.3 kB)
Requirement already satisfied: numpy in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from pyg4ometry) (1.24.4)
Collecting pandas (from pyg4ometry)
  Downloading pandas-2.0.3-cp38-cp38-win_amd64.whl.metadata (18 kB)
Requirement already satisfied: scipy in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from pyg4ometry) (1.10.1)
Collecting sympy>=1.7 (from pyg4ometry)
  Downloading sympy-1.13.2-py3-none-any.whl.metadata (12 kB)
Collecting testtools (from pyg4ometry)
  Downloading testtools-2.7.2-py3-none-any.whl.metadata (5.2 kB)
Collecting vtk (from pyg4ometry)
  Downloading vtk-9.3.1-cp38-cp38-win_amd64.whl.metadata (5.3 kB)
Collecting pygltflib (from pyg4ometry)
  Using cached pygltflib-1.16.2-py3-none-any.whl
Collecting mpmath<1.4,>=1.1.0 (from sympy>=1.7->pyg4ometry)
  Downloading mpmath-1.3.0-py3-none-any.whl.metadata (8.6 kB)
Collecting zipp>=3.1.0 (from importlib-resources->pyg4ometry)
  Using cached zipp-3.20.2-py3-none-any.whl.metadata (3.7 kB)
Requirement already satisfied: cycler>=0.10 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from matplotlib->pyg4ometry) (0.10.0)
Requirement already satisfied: kiwisolver>=1.0.1 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from matplotlib->pyg4ometry) (1.3.1)
Requirement already satisfied: pillow>=6.2.0 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from matplotlib->pyg4ometry) (8.0.1)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.3 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from matplotlib->pyg4ometry) (2.4.7)
Requirement already satisfied: python-dateutil>=2.1 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from matplotlib->pyg4ometry) (2.8.1)
Collecting python-dateutil>=2.1 (from matplotlib->pyg4ometry)
  Downloading python_dateutil-2.9.0.post0-py2.py3-none-any.whl.metadata (8.4 kB)
Collecting pytz>=2020.1 (from pandas->pyg4ometry)
  Downloading pytz-2024.2-py2.py3-none-any.whl.metadata (22 kB)
Collecting tzdata>=2022.1 (from pandas->pyg4ometry)
  Downloading tzdata-2024.1-py2.py3-none-any.whl.metadata (1.4 kB)
Collecting dataclasses-json>=0.0.25 (from pygltflib->pyg4ometry)
  Downloading dataclasses_json-0.6.7-py3-none-any.whl.metadata (25 kB)
Collecting deprecated (from pygltflib->pyg4ometry)
  Downloading Deprecated-1.2.14-py2.py3-none-any.whl.metadata (5.4 kB)
Requirement already satisfied: six in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from cycler>=0.10->matplotlib->pyg4ometry) (1.15.0)
Collecting marshmallow<4.0.0,>=3.18.0 (from dataclasses-json>=0.0.25->pygltflib->pyg4ometry)
  Downloading marshmallow-3.22.0-py3-none-any.whl.metadata (7.2 kB)
Collecting typing-inspect<1,>=0.4.0 (from dataclasses-json>=0.0.25->pygltflib->pyg4ometry)
  Downloading typing_inspect-0.9.0-py3-none-any.whl.metadata (1.5 kB)
Collecting wrapt<2,>=1.10 (from deprecated->pygltflib->pyg4ometry)
  Downloading wrapt-1.16.0-cp38-cp38-win_amd64.whl.metadata (6.8 kB)
Requirement already satisfied: packaging>=17.0 in g:\freecad_38634-1.0.0rc1_lp_2.11-windows-x86_64-py38_kiss\fc_local\bin\lib\site-packages (from marshmallow<4.0.0,>=3.18.0->dataclasses-json>=0.0.25->pygltflib->pyg4ometry) (20.4)
Collecting mypy-extensions>=0.3.0 (from typing-inspect<1,>=0.4.0->dataclasses-json>=0.0.25->pygltflib->pyg4ometry)
  Downloading mypy_extensions-1.0.0-py3-none-any.whl.metadata (1.1 kB)
Collecting typing-extensions>=3.7.4 (from typing-inspect<1,>=0.4.0->dataclasses-json>=0.0.25->pygltflib->pyg4ometry)
  Using cached typing_extensions-4.12.2-py3-none-any.whl.metadata (3.0 kB)
Downloading networkx-3.1-py3-none-any.whl (2.1 MB)
   ---------------------------------------- 2.1/2.1 MB 4.3 MB/s eta 0:00:00
Downloading sympy-1.13.2-py3-none-any.whl (6.2 MB)
[... cut ...]
Downloading marshmallow-3.22.0-py3-none-any.whl (49 kB)
Downloading typing_inspect-0.9.0-py3-none-any.whl (8.8 kB)
Downloading wrapt-1.16.0-cp38-cp38-win_amd64.whl (37 kB)
Downloading mypy_extensions-1.0.0-py3-none-any.whl (4.7 kB)
Using cached typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Building wheels for collected packages: pyg4ometry
  Building wheel for pyg4ometry (pyproject.toml): started
  Building wheel for pyg4ometry (pyproject.toml): finished with status 'error'
Failed to build pyg4ometry
```

KeithSloan, 2024-09-15:
Thanks for Update.

Is it worth contacting the maintainers mentioned on this page?
https://pypi.org/project/pyg4ometry/

Syres916, 2024-09-15:
> Is it worth contacting the maintainers mentioned on this page? https://pypi.org/project/pyg4ometry/

For a Mac, all they are going to do is point you to https://github.com/g4edge/pyg4ometry/tree/main/.github/bin and use the [install-deps-macos.sh](https://github.com/g4edge/pyg4ometry/blob/main/.github/bin/install-deps-macos.sh)

With regard to Windows maybe there's someone else who is going to invest the time required.

luzpaz, 2024-09-15:
Looks like the pyg4ometry devs are active https://github.com/g4edge/pyg4ometry

KeithSloan, 2024-09-15:
I note that if I install pyg4ometry outside of FC i.e. native Python that it installs with python 3.10

I also tried python3.8 -m pip install pyg4ometry

Which downloaded a lot of stuff and seemed okay.

Now seems FC is often using older versions of python and some libraries such as gmsh and obviously now pyg4ometry manage to keep more up to date.

When the addon_manager uses pip for pre-reqs should it be trying a specific python version rather than default to just pip?

luzpaz, 2024-09-18:
Note: this ticket is blocking GDML addon from being compatible with FreeCAD v1.0 Release

chennes, 2024-09-18:
I cannot install that Python pyg4ometry package due to missing CGAL, which is probably the underlying error you are seeing:
```
 × Building wheel for pyg4ometry (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [67 lines of output]
      /private/var/folders/v4/gg68t8g97r77jw78dz117ts00000gn/T/pip-build-env-itzkclv4/overlay/lib/python3.11/site-packages/setuptools_scm/git.py:312: UserWarning: git archive did not support describe output
        warnings.warn("git archive did not support describe output")
      /private/var/folders/v4/gg68t8g97r77jw78dz117ts00000gn/T/pip-build-env-itzkclv4/overlay/lib/python3.11/site-packages/setuptools_scm/git.py:331: UserWarning: unprocessed git archival found (no export subst applied)
        warnings.warn("unprocessed git archival found (no export subst applied)")
      *** scikit-build-core 0.5.0 using CMake 3.30.0 (wheel)
      *** Configuring CMake...
      2024-09-18 16:06:20,399 - scikit_build_core - WARNING - libdir/ldlibrary: /Users/chennes/Documents/FreeCAD/.conda/freecad/lib/libpython3.11.a is not a real file!
      2024-09-18 16:06:20,399 - scikit_build_core - WARNING - Can't find a Python library, got libdir=/Users/chennes/Documents/FreeCAD/.conda/freecad/lib, ldlibrary=libpython3.11.a, multiarch=darwin, masd=None
      loading initial cache file /var/folders/v4/gg68t8g97r77jw78dz117ts00000gn/T/tmpcm25thxr/build/CMakeInit.txt
      -- The C compiler identification is AppleClang 15.0.0.15000309
      -- The CXX compiler identification is AppleClang 15.0.0.15000309
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Setting build type to 'Release'
      -- Found Python: /Users/chennes/Documents/FreeCAD/.conda/freecad/bin/python3 (found version "3.11.8") found components: Interpreter Development.Module
      -- Performing Test HAS_FLTO_THIN
      -- Performing Test HAS_FLTO_THIN - Success
      -- Found pybind11: /Users/chennes/Documents/FreeCAD/.conda/freecad/lib/python3.11/site-packages/pybind11/include (found version "2.13.5")
      CMake Warning (dev) at cmake/modules/FindPythonExtensions.cmake:247 (find_package):
        Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
        are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
        the cmake_policy command to set the policy and suppress this warning.
      
      Call Stack (most recent call first):
        CMakeLists.txt:33 (find_package)
      This warning is for project developers.  Use -Wno-dev to suppress it.
      
      -- Found PythonInterp: /Users/chennes/Documents/FreeCAD/.conda/freecad/bin/python3 (found version "3.11.8")
      CMake Warning (dev) at cmake/modules/FindPythonExtensions.cmake:248 (find_package):
        Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
        are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
        the cmake_policy command to set the policy and suppress this warning.
      
      Call Stack (most recent call first):
        CMakeLists.txt:33 (find_package)
      This warning is for project developers.  Use -Wno-dev to suppress it.
      
      -- Found PythonLibs: /Users/chennes/Documents/FreeCAD/.conda/freecad/lib/libpython3.11.dylib (found version "3.11.8")
      -- Found Cython: /private/var/folders/v4/gg68t8g97r77jw78dz117ts00000gn/T/pip-build-env-itzkclv4/overlay/bin/cython
      CMake Error at CMakeLists.txt:37 (find_package):
        By not providing "FindCGAL.cmake" in CMAKE_MODULE_PATH this project has
        asked CMake to find a package configuration file provided by "CGAL", but
        CMake did not find one.
      
        Could not find a package configuration file provided by "CGAL" (requested
        version 5.0) with any of the following names:
      
          CGALConfig.cmake
          cgal-config.cmake
      
        Add the installation prefix of "CGAL" to CMAKE_PREFIX_PATH or set
        "CGAL_DIR" to a directory containing one of the above files.  If "CGAL"
        provides a separate development package or SDK, be sure it has been
        installed.
      
      
      -- Configuring incomplete, errors occurred!
      
      *** CMake configuration failed
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyg4ometry
Failed to build pyg4ometry
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyg4ometry)
```
The command I ran was the exact one the Addon Manager wants to run:
```
.conda/freecad/bin/python3 -m pip install --target /Users/chennes/Library/Application\ Support/FreeCAD/AdditionalPythonPackages/py311 pyg4ometry
```

KeithSloan, 2024-09-25:
Just for a point of info.

Only the other day we had feed back from a Thomas Braunroth see https://github.com/KeithSloan/GDML/discussions/145

He mentions that they are doing more with FLUKA than with Geant4 these days 

The reason I would like to be able to use pyg4ometry is that we could use it to directly export to FUKA and FLAIR.
ie Under the covers we would export a GDML file and thenuse a small bit of python to produce the files.
I already have a test branch, called fluka that implements this, but it does require the library and functions in the library to do the heavy lifting.

I have currently removed the requirement for pyg4ometry as I don't want people trying to use FC1.0 RC and the workbench to get a bad experience.

luzpaz, 2024-10-17:
what is the way forward here ?