FreeCAD/FreeCAD #16530: Scaling the basic planes during zooming

Pendelf, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you select a plane to create a sketch in PartDesign when you zoom in, the planes become very small

https://github.com/user-attachments/assets/0ee224ce-593f-4d85-998f-8250c85392e3

It may be worth making the planes scale depending on the zoom

https://github.com/user-attachments/assets/c6df0682-3750-40c4-8c48-cd98a158c20e

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.28
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.25
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-14:
Could you add this to https://github.com/FreeCAD/FreeCAD/issues/12584 ?

closed 2024-09-14 by maxwxyz

Pendelf, 2024-09-14:
Ok