FreeCAD/FreeCAD #16531: Spreadsheet/Sketcher V1 RC1 - Adding rows in spreadsheet may break sketches

DerMouse, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have made a MWE to try to explain the problem, which is not completely random but not always triggered either.

When creating a parametric model, with a speadsheet, modifying the speadsheet even in a way totally unrelated to the model (adding a new line between existing lines, without changing any value or aliase anywhere) may break sketches. Breaking is usually a sketch attachment to some face that becomes bad, or external references into some sketch that go amok.

A simple model (two bodies, a box and a lid ; I know clearances on the lid are wrong, it's just to show the problem).
[image]

Associated spreadsheet
[image]

Insert an irrelevant line into the speadsheed, between existing lines
[image]

And the lid sketch goes bad
[image]

Note : sometimes, as in the screenshot above, it's enough to just add an empty line. Other times, the problem appears when filling a field in the inserted line (name or value, depends) or only when defining an alias for the new value. And sometimes nothing bad happens. In my MWE, I've also had the first body thickness (defined through one speadsheet variable) going bad (the variable name was not found anymore, despite still being in the spreadsheet).

Removing the "offending new line" quite often solves the problem, but not always.

### Full version info

```shell
OS: Gentoo Linux
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * persistenttoolbars
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Thie is the FCStd file ; change the ZIP extension back to FCStd to use it.
[Test.zip](https://github.com/user-attachments/files/17001592/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-14:
* #14672 
* #15030

luzpaz, 2024-09-14:
Closing as duplicate

closed 2025-01-25 by luzpaz