FreeCAD/FreeCAD #16533: FreeCAD 1.0rc1 - 3D surface not working. OpenCAMlib issues

Harvie, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using 1.0rc1 appimage from github. I wanted to use 3D surface feature in CAM workbench, therefore i've enable "opencamlib features" in preferences->CAM->advanced. Now i cannot use CAM at all, when i try to open it, it gives me this message:

```
Invalid version: 'camotics: /tmp/.mount_FreeCAxeU3rX/usr/lib/libcurl.so.4: no version information available (required by /usr/lib/libproxy/libpxbackend-1.0.so)
1.2'
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAxeU3rX/usr/Mod/CAM/InitGui.py", line 161, in Initialize
    v = parse(r)
        ^^^^^^^^
  File "/tmp/.mount_FreeCAxeU3rX/usr/lib/python3.11/site-packages/packaging/version.py", line 56, in parse
    return Version(version)
           ^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAxeU3rX/usr/lib/python3.11/site-packages/packaging/version.py", line 202, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
```

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-14:
@Harvie is this a regression ?

Harvie, 2024-09-14:
Not sure. But i saw people on youtube using 3d surface in old versions and i wanted to try it, since 1.0 sounded big. So i guess it is...

luzpaz, 2024-09-14:
> Invalid version: 'camotics:  

@sliptonic do you mind weighing-in ?

Harvie, 2024-09-14:
Is it possible to make some unit test that would at least test whether all workbenches can be loaded when all features are enabled? When OCL is enabled, path workbench does not even load... that sounds like it should be easy to test...

sliptonic, 2024-09-16:
I've reproduced the problem but only in the appimage.  It's caused by camotics and not OCL.
Camotics has to be newer than version 1.2.2 and I don't think they ever released a stable version after 1.2.0 which is a shame. 

The current code tries to import the camotics module and tests if a) it doesn't exist  b) if the version is too old.
When running the appimage, it raises an InvalidVersion error.  I have no idea why that happens only on the appimage but I'll make a pull request to make the code more resilient to unexpected errors. 

In the meantime, you can try uninstalling camotics to get around it.

Harvie, 2024-09-16:
I have couple bits to add:

1.  I have camotics installed and functional on my system, but i still don't see any camotics button in CAM workbench (where is it supposed to be?)
2. I understand FreeCAD only launches Camotics as external application therefore i don't see any reason for FreeCAD to crash in case something is wrong with camotics. (as opposed to OCL which is library that gets called directly in the context of FreeCAD process). Am i right?
3. BTW today i've started reviving old jscut simulator https://harvie.github.io/cnc-simulator/ as a standalone app, even modified it to run offline without triggering cross-origin request protection. It's suprisingly fast, maybe bit faster than camotics and definitely faster than freecad integrated simulator, because it uses shaders. maybe we can think of some kind of integration as well (FreeCAD has some HTML viewer right? i wonder if that supports WebGL). Unfortunately it does not support G02 and G03 at this time...

PR #16583 (2024-09-23): CAM Catch other exceptions with Camotics on import.
Fixes #16533
Attempting to import camotics in the appimage raises an InvalidVersion error which we weren't testing for.

This version catches other exceptions and should fail more gracefully.
closed 2024-09-23 by sliptonic