FreeCAD/FreeCAD #16534: Canvas does not move smoothly holding shift (when using Touchpad mode) if Draft Polyline is selected.

leoheck, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Canvas does not move smoothly holding shift (when using Touchpad mode) if Draft Polyline is selected.
The tooltip says Polyline uses Shift to Constrain. This may be the reason. It should use something else when using Touchpad mode.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git) AppImage
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-14:
Can you share an example file and video?
@Roy-043 @Rexbas FYI

Roy-043, 2024-09-15:
There is a preference available to change the mentioned modifier key.
https://wiki.freecad.org/Draft_Preferences#Grid_and_snapping

Closing this.

leoheck, 2024-09-15:
I am not asking to change the shortcut. I am telling that  touchpad mode is not working as it should.

luzpaz, 2024-09-15:
Maybe you're referring to performance degradation that some are reporting for RC1 ?
*  #16426

leoheck, 2024-09-15:
I still want to share a video, I am  not close to the computer right now. I don't use Draft WB regularly, and if I am not mistaken, it happened when using Draft tools like the Polyline.

Roy-043, 2024-09-15:
When using a Draft command the Shift key has a special function. This conflicts with your Mouse navigation settings. You can do two things: use a different Mouse navigation, or change the related Draft preferences.

Try:
Uncheck "Always snap"
Snap modifier = Shift
Constrain modifier = Ctrl
Check "Always snap"

leoheck, 2024-09-15:
I see, but it conflicts with the default Touchpad Mode. This is weird, it's like Freecad is being released with a purposeful bug.

If Touchpad mode is in use the special function of the Shift key of the Draft Wb should be a different key by default.

Roy-043, 2024-09-15:
Please, for crying out loud, at least TRY what I have suggested.

leoheck, 2024-09-15:
I see @Roy-043 you are suggesting to change something. 

I just have to use Freecad as it is. 
I cannot change it. 
I use it on too many computers (and on different users, since I teach other people) to be able to change it. 

I am telling here that Freecad has this bug:
1. In the `Trackpad mode`, the `Shit key` does something.
1. In the `Draft WB`, the `Sift key` is something else.

These 2 items conflict, this is a bug.

Roy-043, 2024-09-15:
> I just have to use Freecad as it is.

You are obviously not doing that because OOTB the Mouse navigation is not Touchpad.

Leo, please be more constructive. If you are not willing to contribute to the triage of this issue it is better to just close it.

leoheck, 2024-09-15:
I want contribute with the triage, of course. The steps you gave was to test this? I am going to reproduce it asap. It will be later, today. Since I have to put the baby to sleep. It will be definitely before yesterday. Cool?

leoheck, 2024-09-15:
> When using a Draft command the Shift key has a special function. This conflicts with your Mouse navigation settings. You can do two things: use a different Mouse navigation, or change the related Draft preferences.
Try:
Uncheck "Always snap"
Snap modifier = Shift
Constrain modifier = Ctrl
Check "Always snap"

Asking to change the "Mouse navigation" is like asking me to use a different hand to control the mouse pointer.

However, today, after the upgrade from `1.0.0RC1.38684` to `1.1.0dev.38728` I did not observe the issue anymore.
I will assume the issue as fixed and I will close this. 

Thanks for your patience and support @Roy-043.

closed 2024-09-15 by leoheck