FreeCAD/FreeCAD #16537: TNP: Adding a hole prior to another breaks the original hole in RC1

sparker0594, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Adding a hole upstream of an existing hole in the parametric list breaks the existing hole.

To reproduce, open the file provided, reverse the direction of 'Hole001' in 'Flywheel Nut body', then make 'Hole' Visible.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Flywheel Nut Topo Error.zip](https://github.com/user-attachments/files/17002841/Flywheel.Nut.Topo.Error.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-14:
Is this the same behavior as in LS3 branch?
@bgbsww FYI, I guess this will always break, right?

sparker0594, 2024-09-14:
> Is this the same behavior as in LS3 branch? @bgbsww FYI, I guess this will always break, right?

Hi @maxwxyz I'm sorry but what is LS3?

maxwxyz, 2024-09-14:
@sparker0594 https://github.com/realthunder/FreeCAD/releases

bgbsww, 2024-09-14:
LS3 appears to detect and rename this case and main does not.  There is a clear TNP bug here.

bgbsww, 2024-09-14:
LS3:
```
14:31:31  <App> PropertyLinks.cpp(390): Flywheel_Nut_Topo_Error#Hole.Profile auto change element reference Flywheel_Nut_Topo_Error#Groove ;Face13;:Hb37,F.Face26 -> ;#4a6;:Hb37,F.Face28
14:31:31  <App> PropertyLinks.cpp(390): Flywheel_Nut_Topo_Error#Sketch005.ExternalGeometry auto change element reference Flywheel_Nut_Topo_Error#Binder ;g6;SKT;:Hb12,E;WIR;:Hb12:4,E;:H,E.Edge6 -> ;g6;SKT;:Hb12,E;WIR;:Hb12:4,E.Edge6
14:31:31  <App> PropertyLinks.cpp(390): Flywheel_Nut_Topo_Error#Sketch006.ExternalGeometry auto change element reference Flywheel_Nut_Topo_Error#Binder ;g6;SKT;:Hb12,E;WIR;:Hb12:4,E;:H,E.Edge6 -> ;g6;SKT;:Hb12,E;WIR;:Hb12:4,E.Edge6
14:31:32  <App> PropertyLinks.cpp(390): Flywheel_Nut_Topo_Error#Sketch007.AttachmentSupport auto change element reference Flywheel_Nut_Topo_Error#Pocket ;#31b:1;:Hb2e,F.Face5 -> ;#5ce;:Hb2e,F.Face5
```

```
Main:

14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex38
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex38
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex38
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex38
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex49
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Vertex49
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Edge56
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Edge64
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Edge57
14:35:26  <TopoShape> TopoShapeExpansion.cpp(1885): unnamed lower element Edge85
14:36:18  <PropertyLinks> PropertyLinks.cpp(447): Flywheel_Nut_Topo_Error#Hole.Profile missing element reference Flywheel_Nut_Topo_Error#Groove ;Face13;:Hb37,F.Face26
14:36:18  Hole: Sub shape not found: Flywheel_Nut_Topo_Error#Groove.?Face26
14:36:23  Hole: Sub shape not found: Flywheel_Nut_Topo_Error#Groove.?Face26
```

It must that failure to find Face26 in the groove that is causing this.  Why is it failing?

maxwxyz, 2024-09-14:
I'll label it as blocker when it's a TNP issue

bgbsww, 2024-09-15:
So, after playing with this a bit, I'm not so sure this is a TNP error and not a general hole in a groove question.

When I try the example in 0.21.2, reversing the hole results in a Recompute failure on Hole001.

Is there another example, or an explanation of what we are trying to do with the Hole001?

sparker0594, 2024-09-18:
Potentially another example here? 
In the attached file, the hole currently set as the tip has been added to the tree.
It has broken the links of the edges used by the chamfer.

[FootBar.zip](https://github.com/user-attachments/files/17045528/FootBar.zip)

@bgbsww does this help/relate?

bgbsww, 2024-09-23:
Pretty sure this was fixed by #16713   Can we get a confirmation? @sparker0594

CalligaroV, 2024-09-24:
@sparker0594 @maxwxyz @bgbsww, I tried the two files shared in this conversation with

```ruby
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38834 (Git)
Build type: Debug
Branch: main
Hash: 490d6c5abc41bb61fa5bf47398df142a8ac2519e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
``` 

And to me the issue looks solved, at least for the file [Flywheel Nut Topo Error.zip](https://github.com/user-attachments/files/17002841/Flywheel.Nut.Topo.Error.zip)
I can't say anything about [FootBar.zip](https://github.com/user-attachments/files/17045528/FootBar.zip) because whenever I open it I'm not able to perform a full recompute without errors (something's wrong in `FootBar->Body->Chamfer`)

@sparker0594, can you share that file before the operation that causes the error?
Or try it yourself and give us feedback if after #16713 the issue looks solved also for you?

TIA!

sparker0594, 2024-09-25:
@bgbsww @CalligaroV 

I have tested both files and it seams to be fixed.
@CalligaroV in the [FootBar.zip](https://github.com/user-attachments/files/17045528/FootBar.zip) file the chamfer was broken due to this error, ad could not be recovered. after removing and re-creating in the same step i did originally the file compiles as expected, the chamfer does not break!

I consider this closed, thank you for your efforts everyone!!!

closed 2024-09-25 by sparker0594