FreeCAD/FreeCAD #16538: TechDraw: Two PDF files are created when using virtual PDF printer

semhustej, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When printing TechDraw page (using 'Print....' command or 'Print preview..,' command) using virtual PDF printer (I tried with Microsoft Print to PDF and PDF Creator), two PDF files are created. One is the correct print, the other one is empty PDF file.

I am not able to reproduce this behavior with FC 0.21.2, so it seems like a regression. I am not able to reproduce this problem when using 'Print...' command from main 3D view.

My test file (I could reproduce with 2 files):
[techdraw-print-test.zip](https://github.com/user-attachments/files/17002972/techdraw-print-test.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-14:
@WandererFan FYI

xtemp09, 2024-09-14:
I have just printed the applied example file with _PDF24 Creator 11.18.0_, _Foxit PhantomPDF 10_ and _Microsoft Print to PDF_.

- the first PDF file is empty, the second is not ―  1.0.0RC1.38642 (Git)
- an only file (second) ― 0.21.2.33771 (Git).

<details><summary>0.21.2.33771 (Git)</summary>

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>1.0.0RC1.38642 (Git)</summary>

```ruby
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

</details>

redPanther, 2024-09-15:
I tested it im Linux (kubuntu 24.04, kde). Did "print" and then selected the "print in pdf file" in the printer dialog. The result is one pdf file wit correct content - so at least for me no issues for pdf printing on linux/kde.

semhustej, 2024-09-15:
Thanks for testing. Looks like a Windows issue.

WandererFan, 2024-09-15:
I can not duplicate this on my linux box.   Is the behaviour different if you use the "Export to PDF" option from the RMB context menu?

I'll see if I can get my windows vm up for a test.

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38740 +1 (Git)
Build type: debug
Branch: winFilespec
Hash: 78c1a110f4c6ee0b69444f5a821c0b83b2fa8a14
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

semhustej, 2024-09-15:
When using 'Export PDF' from the RMB context menu, only one PDF is created - which is OK. This is the same behavior as when using File -> 'Export PDF...' tool.

The bug only happens when virtual PDF printer is used through the 'Print...' command.

benj5378, 2024-09-16:
Could this be template related? @semhustej can you recreate on any shipped templates? I wondering if the weird corners could trigger a second page.

semhustej, 2024-09-16:
> Could this be template related? @semhustej can you reproduce on any shipped templates? I wondering if the weird corners could trigger a second page.
> 
> Can you reproduce when setting printer properties margin to 0?
> 
> I'm not even sure how the printer is triggered twice in two separate files? What are the names of file 1 and file 2?

It is not template related - I tried different ones from the default FC templates. I tried with A3 sized templates and A4 sized templates. I am also able to print from FC 0.21.2 without creating two files.

It does not seem to me, that margins are the problem. Even if I print blank page, the output is still double file.

I recorded a  video showing how two files are created:
https://1drv.ms/v/c/d058e155a9fc1abb/EfxAFrykslNNoXjFAcUxsS0BjXZyrh8OgpV5bI8c4Oq6pw?e=E8rgYS

benj5378, 2024-09-16:
@semhustej, the video helped a lot. I see that the printer dialog comes up twice. However, then I'm even unsure if this is a TechDraw issue... I do notice, that Linux uses a Qt print dialog, while Windows uses the built-in Windows print dialog.

Are you able to compile FreeCAD? If yes, can you put a break point at `MDIViewPage::printAllPages` and do a backtrace for both times the breakpoint is reached? I want to know from where the function is called twice.

https://github.com/FreeCAD/FreeCAD/blob/758674d40be45e365f4148fb2a5ea6bd6f93bc74/src/Mod/TechDraw/Gui/MDIViewPage.cpp#L384

semhustej, 2024-09-16:
Hi @benj5378,

yes, video is always helpful - sometimes when describing the issue one omits an important detail.

I am afraid compiling the software from source code is out of my abilities.

I figured it is a TechDraw related issue, because when printing selected elements from the main 3D window, only one PDF file is created. Not sure which other workbench I could test with printing to make it clear or disprove that it is TD related.

Syres916, 2024-09-18:
> Not sure which other workbench I could test with printing to make it clear or disprove that it is TD related.

Spreadsheet would be the more obvious choice, no?

semhustej, 2024-09-18:
> Spreadsheet would be the more obvious choice, no?

I tested printing spreadsheet using both Microsoft Print to PDF and PDF Creator. Both create just 1 PDF file, which is correct.

I additionally  tried using virtual printer 'Microsoft XPS Document writer' and that too creates 2 files when printing a TechDraw page.

It looks like a combination of TechDraw page + Print command + Virtual printer on Windows.

closed 2024-09-23 by chennes

kisolre, 2024-09-23:
I had an issue that when I print a Techdraw page I always get an empty one printed after the page with the drawing. Did not print too much to bother me enough to report. Lets see if that also fixes this issue.

commit d5d66b2c (2024-09-23): [TD]fix double pdf file on win (fix #16538)
commit b99e34bd (2024-09-23): [TD]fix double pdf file on win (fix #16538)