FreeCAD/FreeCAD #16541: Multifusion failed

cropduster32, 2024-09-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a union of 2 solids, either from the menu or the icon, it fails with the following msg in report view:

21:55:57  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
21:58:45  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\cropd\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Part\BOPTools\BOPFeatures.py", line 88, in make_multi_fuse
    self.copy_visual_attributes(obj, obj.Shapes[0])
  File "C:\Users\cropd\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Part\BOPTools\BOPFeatures.py", line 106, in copy_visual_attributes
    target.ViewObject.ShapeAppearance = source.ViewObject.ShapeAppearance
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'Gui.ViewProviderLink' object has no attribute 'ShapeAppearance'
21:58:49  <Exception> MultiFusion failed
21:58:49  Fusion: MultiFusion failed

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.45
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

[Archimedes-Screw.zip](https://github.com/user-attachments/files/17003815/Archimedes-Screw.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-15:
@cropduster32 how do you create the fusion? Please add steps how to reproduce.
@davesrocketshop seems to be an issue with `ShapeAppearance`

Roy-043, 2024-09-15:
Steps to reproduce:
Open the file.
Delete the existing Fusion.
Select the array 1st.
Add the Sweep to the selection.
Press the Part_Fuse button.

The problem does not occur if you select the Sweep 1st.

Note: the fuse operation itself fails for a different reason. The blades of the screw should be solids instead of just faces.

PR #18762 (2025-01-06): Part: copy_visual_attributes did not handle Draft Link arrays
Fixes #16541.
Fixes #18760.
commit af667d6f (2025-01-06): Part: copy_visual_attributes did not handle Draft Link arrays

Fixes #16541.
Fixes #18760.
closed 2025-01-06 by yorikvanhavre