FreeCAD/FreeCAD #16543: DXF import (C++): cannot import as Draft objects

furgo16, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using FreeCAD for manipulating DXF drawings (e.g. with the Draft or BIMi workbenches, for architectural models), arguably one of the most useful legacy (Python) DXF importer features is the ability to import DXF entities as Draft objects. As opposed importing entities as Part shapes, which is also valid use case supported by both the legacy and new importer.

The new ( C++) DXF importer does not have this ability and can only import entities as Part shapes.

Implementing this could be a major part in the puzzle of achieving feature parity with the legacy importer. And eventually replacing it for the benefit of improved usability and code maintenance.

### Full version info

```shell
1.0RC
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-15:
@Roy-043 @kpemartin FYI

Roy-043, 2024-09-15:
This is a known issue. I think you can 'draftify' the results of the C++ importer. Either by using Draft_Upgrade, or with some Python code.

kpemartin, 2024-09-15:
Indeed, when I restructured the C++ importer (part of resolving #7551) I included but did not implement the ability to do draft-object importing, and before implementing this I was distracted by life.
I must admit that I also mentally made it a lower priority than some of the other issues I'd found (but not necessarily reported). My familiarity with FreeCAD itself (as a user) is still limited and I'm a little fuzzy on the relative benefits of shapes vs. Draft objects. Draft objects are custom Python objects that have properties relevant to the actual shape (e.g. a line's endpoints, center and diameter for a circle, etc.) which are difficult to extract, and even harder to modify, when shapes are used, even when not compounded. On the other hand the Python Draft objects come at a cost in terms of memory use, processor time, and save-file size, and are created whether or not they are needed for any particular object.
It is probably not in the cards, but my approach to this would be to provide Python manipulator objects (different classes for different shapes) which are only created when needed. Each class would have properties (e.g. center point and radius) and perhaps methods (e.g. break a circle into two arcs) appropriate for the shape. You would create one appropriate for the actual shape, giving the shape object to the constructor (which would verify that the shape is appropriate), do the inspections/modifications, then lose the manipulator's reference (or keep it if you really want to)

furgo16, 2024-09-15:
Thank you. I can only speak for myself, and as someone with limited knowledge. In any case, here are my comments:

1. Regarding the workaround: indeed, you can upgrade shapes into draft objects, which I do when I can. There are some caveats: you need to do it one by one, otherwise multiple selected shapes get lumped into a block. This is feasible for a handful of shapes, but it is no longer when you have a few dozens or more, which is not uncommon in architectural drawings. Good point about doing this in Python (I'll try that next and report back). Also, for some reason, sometimes some of the shapes created by the importer cannot be upgraded (using Draft_Upgrade does nothing). 
1. Regarding the benefits of shapes vs. Draft objects: to me, it's simply that you can manipulate draft objects much more easily with the Draft and BIM workbenches. My limited experience might be betraying me, but for that use case (modifying architectural drawings and creating BIM objects from them) 2D shapes feel rather limited in comparison.

The ability to import Draft objects is the main reason I've been using either the legacy importer or [ezydxf](https://forum.freecad.org/viewtopic.php?t=81962&start=20), but that comes at a cost when it takes 20 minutes or more to import a reasonably sized DXF file. That's why I'm excited about the C++ importer going in the direction of feature parity with the Python one. Ultimately also because having two importers is a very confusing experience for new users. I'm stating this as honest feedback, in no way dismissing the great work and efforts that have been put in both importers.

kpemartin, 2024-09-15:
Adding the creation of Draft objects to the C++ importer is essentially grunt work, generally starting with what the legacy importer does and writing C++ code to do the equivalent thing, which generally amounts to preparing a bunch of arguments from the DXF data and calling `Draft.make_xxx`. Some objects also require fiddling with properties afterwards.

If there is a general consensus that this is particularly important I can start working on that as a bit of a background task.

furgo16, 2024-09-15:
@yorikvanhavre thoughts about that? (adding support for Draft object creation to the C++ importer)

yorikvanhavre, 2024-09-23:
It's IMHO an annoying thing to add to the C++ DXF importer. As much as it's easy for Python modules to work with C++ code, the contrary is tedious to do and does not give very clean or solid code...

It would be I think much more interesting to have a macro or tool to convert just ANY 2D compound to Draft entities...

Such a macro could basically:

1. Check that the object is flat. No solids, and bounding box's Z dimension is zero
2. Separate all faces
3. Separate all remaining wires
4. Separate all remaining edges
5. Analyze which face is a rectangle, convert them to a rectangle with makeFace = True
6. Analyze which wire is a rectangle, convert them to a rectange with makeFace = False
7. Analyze which face is a circle, convert them to a circle with makeFace = True
8. Analyze which wire is a circle, convert them to a circle with makeFace = False
9. Convert remaining faces and wires to polylines
10. Convert remaining edges to lines or arcs

kpemartin, 2024-09-23:
I plan on making the DXF importer work properly for arbitrarily-oriented 2D objects so step (1) would make this an insufficient solution.

I actually took a stab at making the importer use Draft objects last week and decided I didn't want to make the change with so many pending merges in the pipeline. It didn't seem that difficult to do, though I must admit I did a simple case (LINE). Getting 2D objects (ARC, etc) to orient properly would be more work, but then these don't import properly in general anyway.

furgo16, 2024-11-16:
@kpemartin now that the merge queue is more stable and post-1.0 MRs are being merged, do you think it would be a good time to resome this work? Thanks!

kpemartin, 2024-11-17:
I'm trying to wrap up my fix for #16849 right now. It is all working but I have to make the changes presentable and also check for any new complaints from all the CI checks. As part of this I've also found some other performance notes that I have to turn into Issues.

I think I have a have a basis for fixing this Issue sitting in a stash somewhere, leaving mostly the grunt work of making all the python calls to create the objects and set all the new objects' properties

maxwxyz, 2025-01-22:
@Roy-043 should we close this?

kpemartin, 2025-01-23:
I don't care much either way; I'm going to use #11875 as the main Issue to address.
The advantage of keeping it open is that once I get the PR to fix it (and the also-similar #13598) the PR can be marked as fixing all three so they won't be mysteriously "resolved", but will instead reference a proper fix,