FreeCAD/FreeCAD #16545: SubShapeBinder object is not displayed in the model tree

Zehrondit, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you create a SubShapeBinder object from any face of the model, it does not appear in the viewport or in the model tree. At the same time, this object is present in the body property “group”.

If you open the file in an old version of FreeCAD everything is displayed normally. 

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-15:
I can't reproduce. This may be related to Tree view settings and/or theming?

luzpaz, 2024-09-15:
@Zehrondit can you install weekly build and see if you can reproduce ?  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Zehrondit, 2024-09-15:
@Roy-043 Suspect the problem is on windows, at least I don't have access to linux right now, can't check.
@luzpaz Yes, I have the same problem on the latest daily version as well
[image]

Roy-043, 2024-09-15:
I am using windows. I can't reproduce on the FreeCAD version listed below. Nor on previous weekly versions (including 38643).

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
Installed mods:
```

Zehrondit, 2024-09-16:
@Roy-043 
I'm using Windows 11, maybe that's the problem?

Roy-043, 2024-09-16:
That seems unlikely. As mentioned before, I suspect Tree view settings or theming. Can you try to reproduce the issue with a reset CFG and the default theme?

Zehrondit, 2024-09-16:
After a reset in the settings I got this:

`15:41:37  <Exception> FreeCAD Exception
15:41:37  0 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0004317 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0009297 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0013573 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0017668 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0022639 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0029959 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0034314 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0039636 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0046961 Tree.cpp(3841): view provider not attached
15:41:53  ~Subject(): Not detached all observers yet
15:41:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
`

Zehrondit, 2024-09-16:
After a reset in the settings I got this:

```
15:41:37  <Exception> FreeCAD Exception
15:41:37  0 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0004317 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0009297 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0013573 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0017668 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0022639 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0029959 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0034314 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0039636 Tree.cpp(3841): view provider not attached
15:41:37  <Exception> FreeCAD Exception
15:41:37  0.0046961 Tree.cpp(3841): view provider not attached
15:41:53  ~Subject(): Not detached all observers yet
15:41:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

luzpaz, 2024-09-17:
Another Windows 11 Access Violation ?  
https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD+Access+violation&type=issues

Zehrondit, 2024-09-21:
After reading this thread on Windows 11, I did what it said: renamed the configuration files so that they would not interact with the program
I also unzipped the program to the root of the C drive, not to the desktop as before.
I'm not sure if changing the configuration helps, I suspect it's the Cyrillic alphabet in the path name.

closed 2024-09-22 by Zehrondit