FreeCAD/FreeCAD #16547: Negative values for joints

Huwyler, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I create a rack and pinion joint, it often happens that the direction is wrong. The intuitive solution for this is to just add a negative value for the pitch radius. However, this is not possible, as soon as I hit "-", I cannot enter a number any more.

However, in the properties editor, it is possible and works just as expected.

This probably also applies to others like screw or gear/belt joints.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-wayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.28
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-15:
We need more info? Are you working within a certain workbench? Please elaborate so we can reproduce.

Huwyler, 2024-09-15:
Ah, sorry. I mean the rack and pinion joint in the new assembly workbench. 
Steps to reproduce:
Create an assembly
Add a gear and and a rack
Create a rack and pinion 
Add a rack and pinion joint
Try to add a negative value to the radius

Huwyler, 2024-09-15:
Sorry, I mean the new assembly workbench.

Steps to reproduce:
Create an assembly with a gear and a rack as a preparation for a rack and pinion setup

Create a rack and pinion joint

Try to set the radius value to a negative mumber. Not possible

Finish the joint (with a positive radius)

Change the radius to a negative number in the object editor. Works, and changes the "direction" of the joint

xorxx, 2024-12-04:
I can confirm this bug. Changing the formula for distance in the property editor to a negative sign does fix the rotation direction. Doing the same in the task editor does not do the job. The formula is correctly saved but the sign is lost somewhere in the calculation path.