FreeCAD/FreeCAD #16549: In Sketcher link actions menu keeps popping up when selecting points

mccarthyam, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am trying in sketcher to select two points as add a constraint but the link actions menu keeps popping up preventing me from adding the constraint 

See attached screen video of the issue

https://github.com/user-attachments/assets/5d73059c-b581-4576-8011-a828277f4f04

### Full version info

```shell
1.0.0RC1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-15:
@mccarthyam full About info, please

mccarthyam, 2024-09-15:
Please find about infor
[image]

Roy-043, 2024-09-15:
I suspect this is [Mouse navigation](https://wiki.freecad.org/Mouse_navigation) related. Please check if the issue occurs with the "CAD navigation" option.

mccarthyam, 2024-09-15:
Tried it with the CAD navigation option and the problem does not re-occur

Roy-043, 2024-09-16:
Which mouse navigation do you use?
Do you hold down the Ctrl key (which is not necessary) when selecting elements in sketch edit mode?

mccarthyam, 2024-09-20:
I used the gesture navigation. I changed to TinkerCad and the issue has not re-occured

luzpaz, 2024-09-20:
@Rexbas another gesture navigation issue, have you seen this before ?

Rexbas, 2024-09-20:
Yes, I have seen similar behavior in other navigation styles and I have fixed them some time ago. I will see if I can also fix this too.

Rexbas, 2024-10-20:
I cannot reproduce this. @mccarthyam, could you please describe in detail what you are doing. For example: press LMB, move mouse, release LMB. And are you using a mouse or a laptop touchpad?

luzpaz, 2024-10-20:
@Rexbas can you add the [About](https://wiki.freecad.org/About) info of the version you tested? TIA

@mccarthyam can you test on latest weekly build and report back? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) TIA

Rexbas, 2024-10-20:
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.0, Qt 5.15.8, Coin 4.0.3, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.23