FreeCAD/FreeCAD #16550: DXF importer (Python): geometry is imported individually even if "Group layers into blocks" is selected

furgo16, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download the [sample DXF file](https://github.com/jscad/sample-files/blob/master/dxf/dxf-parser/layers.dxf)
1. Start FreeCAD
1. Create a new document
1. File > Import
1. Open the DXF sample file from step 1.
1. In the preferences dialog, use the import preferences from the screenshot below. Most importantly, tick `Use legacy python importer`, `Use layers`, and `Group layers into blocks` .
1. **Expected**: DXF entities are imported as a block by layer
1. **Actual**: DXF entities are imported as a block by layer, but individual entities are also imported as individual objects, marked as hidden (see screenshot).

I think what happened here is that accidentally the Create: `Draft objects` option was chosen. `Draft objects` and `Group layers into blocks` should probably be mutually exclusive.

[image]

[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38720 +8 (Git)
Build type: Unknown
Branch: Issue16068
Hash: 6149c0c015fab15ad6996d3060c6af103de59337
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-15:
@kpemartin @Roy-043 FYI

kpemartin, 2024-09-15:
Only "simple Part shapes" can be combined into compound shapes, Draft objects cannot.

My feeling it that the "group layers into blocks" and also "join geometry" (which I understand, perhaps incorrectly, to mean "group objects that connect end-to-end into blocks") should just be more radio-button choices along with "create...Draft objects" and "create...Simple Part Shapes" as these are more or less mutually exclusive and represent different degrees of simplification of the DXF contents (although "create..Sketches" may not fit this model, also why is "Sketches" plural... doesn't this always just create one Sketch?).

A bit of reading of the legacy importer code might reveal whether this mutual exclusivity is already de facto true.

Some of the option text could be re-worded for clarity, and each option would be implicitly "as much as possible" so for instance Dimensions would always be Draft objects and never combined.

Roy-043, 2024-11-25:
IMO the problem is mainly that the Draft_Block object does not claim its children. In #18129 a Part_Compound is used instead. The individual Draft (or Sketch) objects then appear nested. This makes more sense.
[image]

PR #18129 (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior
Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
commit 7a7c042a (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior (#18129)

Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
commit 9d38f0e8 (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior

Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
closed 2024-12-04 by Roy-043