FreeCAD/FreeCAD #16551: Materials and Colors

leoheck, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Material thing seems still to be a broken feature. Is it finished already?

This widget was added in a bad way. 
1. The material widget should start collapsed by default (it is kind of useless, and takes a lot of space)
1. The input field does not look and find a material on that huge list (it eclipses the "Custom appearance" button)
1. I cannot set color properly anymore, colors do not change on the canvas.
1. I have to click 3 things to reset material/colors. 

Similarly `Set appearance per face` seems not to be working. Something is off.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-09-15:
Also, when using Themes, it looks like it is a bit hard to select the expand/collapse button to improve it without messing with the other buttons. https://github.com/obelisk79/OpenTheme/issues/121

maxwxyz, 2024-09-15:
again you create poor issues. I'm closing this for multiple topics and some duplicates.

closed 2024-09-15 by maxwxyz

leoheck, 2024-09-15:
I am working, @maxwxyz (on Sunday). I don't have too much time to make an extended survey, I am sorry.
I could also not find related issues. 

It looks like you have a better knowledge of the issues base.
Do you have references to share? You have shared none to base what you are saying. 

This Material and Color feature was poorly created and it regressed the previously functionality. It is currently not useful.