FreeCAD/FreeCAD #16552: DXF importer (Python): blocks are imported outside their layers

furgo16, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download the [sample DXF file](https://github.com/jscad/sample-files/blob/master/dxf/dxf-parser/layers.dxf)
1. Start FreeCAD
1. Create a new document
1. File > Import
1. Open the DXF sample file from step 1.
1. In the preferences dialog, use the import preferences from the screenshot below. Most importantly, tick `Use legacy python importer`, `Use layers`, Create `Part shapes` and `Group layers into blocks` .
1. **Expected**: DXF entities are imported as a block by layer
1. **Actual**: DXF entities are imported as a block by layer, but the blocks are outside of the layers (see screenshot)

Also note the naming of the blocks: the block that should be under the `0` layer is named `_`. I believe this is because there is already an object with the "0" label (the layer), so when the importer creates the child block with the same "0" label, FreeCAD renames it. The renaming logic seems to be to replace the first digit of the label by an underscore, if the label starts with a digit. A workaround is to prepend a letter to the label. Looking at the output of #16550, it seems that the legacy importer works around this by prepending `block` to the created block, but for some reason that is not happening here. 

[image]

[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38720 +8 (Git)
Build type: Unknown
Branch: Issue16068
Hash: 6149c0c015fab15ad6996d3060c6af103de59337
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-15:
@kpemartin @Roy-043 FYI

kpemartin, 2024-09-15:
This seems to be asking for the Legacy importer to back-implement behaviour of the C++ importer, which leaves the compound objects within the layers since there can be several such compound objects (if, say, the colours are different), and there can be non-compoundable objects (like dimensions).

I do not believe that the legacy importer checks attributes and always merges all the shapes of the layer into a single compound, so it is a bit more justified to just place this compound at top level instead of in the layer. I'm not sure what it does if there are also dimensions or annotations in the layer.

furgo16, 2024-09-16:
> so it is a bit more justified to just place this compound at top level instead of in the layer.

Thanks for the detailed analysis. I'm not familiar with the implementation details, but from the user perspective, I would think the expectation would be to have the blocks under the layers.

The workaround is to move the blocks manually under the layers after the import, but I feel a more optimal experience woul be for this to happen automatically like in the C++ importer.

> I'm not sure what it does if there are also dimensions or annotations in the layer.

I can test this in the next few days and report back.

furgo16, 2024-09-16:
I added a text line and a dimension to the sample file with QCAD. I then imported it with the legacy importer and settings in the screensheet at the bottom

Result:
- The legacy importer imports the text and dimension as individual (Draft?) objects, inside the corresponding layer. They are not lumped into one of the imported blocks.

[image]

[image]

Roy-043, 2024-11-25:
For posterity a DXF with a text and a dimension.

[layersr14.dxf.txt](https://github.com/user-attachments/files/17907725/layersr14.dxf.txt) (remove the .txt extension).

PR #18129 (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior
Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
commit 7a7c042a (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior (#18129)

Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
commit 9d38f0e8 (2024-12-04): Draft: Legacy DXF importer: fix layer block behavior

Fixes #16550.
Fixes #16552.

* Layer blocks receive a name with the "LayerBlock_" prefix.
* They are put in a layer if the "Use layer" import option is checked.
* If Draft or Sketch objects are created a Part_Compound instead of a Draft_Block is created. This was done because a Draft_Block does not claim its children. I am not sure about this change. Maybe the behavior of the Draft_Block should be changed instead.

@yorikvanhavre: do you know why a Draft_Block does not claim its children?
closed 2024-12-04 by Roy-043