FreeCAD/FreeCAD #16555: New Feature: Task commands (Ok/Cancel) Widget

leoheck, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a small display (15" laptop).

The Task widget most of the time shows as scrollbar since the screen size is not big.
This is fine, but the main buttons Cancel/Ok are close to the top, and they are always hidden when finishing configuring the Task. Then you have always to scroll back up again to find those control buttons. The results message of the Task could also be displayed there. Saying the user for those annoying and disrupting popups. 

My suggestion is to have a new widget with the Task controls. So I can stack them to have them always visible somewhere.

When this widget is displayed, the Task widget should not show the controls of course.
When there is no task open, it should maintain the size, messages could be displayed there too. 

### Full version info

```shell
1.0-RC1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-16:
@leoheck, could you point out an example application having this UI concept?

FlachyJoe, 2024-09-16:
I agree it could be useful. The OK/Cancel buttons could remain on the toolbar even when the Task panel auto-hides.

xtemp09, 2024-09-16:
@FlachyJoe, what application utilizes this concept?

leoheck, 2024-09-16:
> @leoheck, could you point out an example application having this UI concept?

Unfortunately I don't have any reference to share. This is my original idea 😅 . I wanted to have a better user experience while not destroying the user experience of the others that like the Task controls where they are. 

So, if this could be implemented somehow, it would not affect the default usage. It would only improve, incrementally for the people that think the current Tasks widget is not efficient. 

Having the Tasks control detachable is an interesting concept since the user can decide where to place it. On top, or at the bottom of the Task panel would be my initial guess.

leoheck, 2024-09-16:
Maybe it is not called window controls. It is more like a widget with buttons. I will try to draw this somehow.

xtemp09, 2024-09-16:
> Having the Tasks control detachable is an interesting concept 

> Maybe it is not called window controls. It is more like a widget with buttons. I will try to draw this somehow.

It is not clear what you want, you are might possibly thinking of Floating UI, but Floating UI on PC is used for functions that can be employed in the beginning of the work with a document (Zoom In, Zoom Out) or rarely used context functions (Measure Distance, Scroll to the top, text style modification etc.). On _Android_ it is used for everything to save some screen space.

Another concept is rollouts, like _3ds Max_ and _Blender_. But it is used for multitude of options and parameters _FreeCAD_ usually lacks in tasks. It is possible to utilize this concept only in options.

leoheck, 2024-09-16:
Sure, it is not clear because it is just an idea that can be brainstormed and I am not completely aware of the name of the features of the interface.

This image shows what I am trying to describe.
Here, the `Cancel` and `Ok` buttons were removed from the main Task panel and moved to another separated widget (where I called Task Controls).
So I can scroll the Task, and the `Cancel` and `Ok`buttons will remain be always visible.

[image]

For instance, the current behavior is this one.
The Task widget has more constant than it can display at once, most of the time (specially for small displays).
Then a scrollbar appears.
So, to configure the Task, the user has to fill the items there (generally following form the top to the bottom), and at the end, most of the time, the view is at the bottom, where the `Cancel` and `Ok` buttons are hidden.
Then the user has to scroll everything back up again to accept/cancel the Task settgins.
Scrolling the Task panel is also a poor thing that will be improved here, because you have to place the cursor at some specific areas otherwise it can be placed on top of some things that capture the scroll of the whole Task... like lists. And sometimes we mess the configuration (without even noticing it) just because the selected item was scrolled to another option. This idea would reduce the frequency of this issue too.
So, by having a separated widget/panel/window for the `Cancel` and `Ok` buttons would solve this problem without messing with the current flow/layout for the ones that like the way it is now.

[image]

leoheck, 2024-09-16:
> I agree it could be useful. The OK/Cancel buttons could remain on the toolbar even when the Task panel auto-hides.

Yeah, this would help, but some users could complain. So having a possibility of having them in another widget could improve this without annoying others.

luzpaz, 2024-09-16:
Related small screen ticket: 
* #12742

xtemp09, 2024-09-17:
The simplest solution is to place everything but _Ok/Cancel_ buttons within a scroll area. The buttons will remain in place on top of the task all the time, while everything else can be scrolled.

This is used pretty much everywhere in _Windows_ from standard messageboxes to settings (for example, _Microsoft Word_), though, in this case _Ok/Cancel_ buttons are placed at the bottom of the window.

Its reimplementation seems to be incredibly simple, but requires bravery of maintainers.