FreeCAD/FreeCAD #16559: mouse cursor disappears when Draft Line is selected

Laukku, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After selecting the Draft workbench, and clicking any of the draft tools (line, rectangle etc) the mouse cursor is not visible in the 3D view until I click for the first time, thus being only able to start any draft action from a random point. The cursor becomes visible once clicked for the first time, so the shape can be finished at a specific point. The cursor is visible everywhere else, only disappearing when moved to the 3D view window. If the 3D view is a separate window, the cursor is visible; the problem only happens when the window is docked.

### Full version info

```shell
Also happens in 0.21.2, but tried downgrading to see if older versions work
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Finnish/Finland (fi_FI)
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

Fresh install on Windows. New Project. Change to Draft workbench. Click any draft tool.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-16:
Cannot reproduce this (Debian testing with current code in main branch).

- Pressing Draft Line tool
- The cursor stays visible
- When moving the mouse over the 3D view, the cursor becomes a crosshair (the "detail" cursor).

What happens if you have objects that can be snapped? Do you see the snap symbol?

xtemp09, 2024-09-16:
In the issue #14570 it was graphics drivers or some tweaks.

I saw the source code, there is no difference between the "docked" and "undocked" states for 3D view. When user "(un)docks" the view is cloned (recreated in a window, the window can be separate or not).

Laukku, 2024-09-16:
@maxwxyz Also happens in dev version.
@yorikvanhavre I am assuming a draft line could be snapped to. In that case, if the mouse is on the line, it turns yellow but otherwise there is no indication of snapping, even if I try to find the end point of the line.

I was able to use the Sketch and Part workbenches just fine in all versions I've tried so far.

semhustej, 2024-09-18:
I am unable to reproduce the issue. Using notebook with Intel/NVIDIA graphics card.

`OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
`

luzpaz, 2024-09-18:
@Laukku please paste your full [About](https://wiki.freecad.org/About) info for dev version you tested on. TIA

luzpaz, 2025-07-23:
@Laukku still relevant ?

semhustej, 2025-07-28:
> While testing this I notice a different(?) but perhaps related issue:
> 
>     1. Switch off snapping.
> 
>     2. Start the Draft_Wire command.
> 
>     3. The cursor in the 3D view is a cross.
> 
>     4. Pick the first point.
> 
>     5. The cursor is now an arrow.

I can confirm this behavior. It happens with all draft objects (line, circle, rectangle, etc.). BIM objects always use cross when asking for next point (tested with Wall, Beam, CurtainWall).

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42721 (Git)
Build date: 2025/07/25 17:44:16
Build type: Release
Branch: main
Hash: b5d782e85690f12be44513e1f1e510b367a69c27
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```