FreeCAD/FreeCAD #16560: Model view performance degradation when using Qt6 (Additive Helix)

mplzik, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating a fairly simple AdditiveHelix Body in Freecad 1.0rc1 built with Qt6 enabled, the UI becomes quite sluggish when moving mouse over the Body and waiting for edges/surfaces to be highlighted. Some Other operations (moving or rotating the view) seem to be unaffected. When using locally buildt FreeCAD 1.0rc1 that used Qt5, I was not able to reproduce the problem

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_macros
  * Curves 0.6.46
  * Render 2024.8.4
```

### Subproject(s) affected?

PartDesign

### Anything else?

Model used to trigger the issue:
[meteo2.zip](https://github.com/user-attachments/files/17007057/meteo2.zip)

Video of the issue:
[Screencast from 2024-09-14 21-21-32.webm](https://github.com/user-attachments/assets/70b5fc31-7f2c-4f08-a862-14e8d15fdd77)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-09-15:
Perhaps, this is feasible to introspect with _GammaRay_.

maxwxyz, 2024-09-16:
@FlachyJoe FYI

jffmichi, 2024-09-17:
I may have noticed a similar issue but only when running FreeCAD natively on Wayland. Not sure if your setup is using Wayland but maybe you could try running on XWayland using
```
QT_QPA_PLATFORM=xcb FreeCAD
```
and check if the issue still happens.

mplzik, 2024-09-17:
Indeed, I'm running FreeCAD on native Wayland. I just tested launching FreeCAD using the command you provided and the performance was back to normal.

GRFTSOL, 2024-12-13:
> I may have noticed a similar issue but only when running FreeCAD natively on Wayland. Not sure if your setup is using Wayland but maybe you could try running on XWayland using
> 
> ```
> QT_QPA_PLATFORM=xcb FreeCAD
> ```
> 
> and check if the issue still happens.
This fixed for me.
Running on archlinux+wayland

ymndoseijin, 2025-03-04:
I can confirm this is still an issue on Arch Linux using Sway on FreeCAD 1.0.0, and that forcing XWayland fixes it. In my case, I was having problems using an Additive Pipe, this is the link to the [forum post](https://forum.freecad.org/viewtopic.php?p=814649#p814649) I made.

jffmichi, 2025-07-23:
still exists -.-

FEA-eng, 2026-09-07:
@mplzik Can you confirm if it's still relevant?

mplzik, 2026-09-07:
Hello there, reporting on the following version of FreeCAD>
```
FreeCAD 1.1.3, Libs: 1.1.3R44987 (Git)
(C) 2001-2026 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
```

The lagging improved significantly -- when moving the cursor around the same model, I only see ~500ms latency between mouse cursor being above a surface and the surface getting highlighted; the name of the face/edge in the status bar refreshing even faster.  Notably, no "application is not responding" screens anymore, although the 500ms lagging is still a bit annoying.

Also, it looks like the workaround mentioned previously (`QT_QPA_PLATFORM=xcb`) does not have any effect on lagging.