FreeCAD/FreeCAD #16561: ability to enable and disable navigation cube

Roberto8303, 2024-09-15:
### Original PR

https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=needs+triage%2Cbackport&projects=&template=BACKPORT_REQUEST.yml&title=%5BBackport%5D+

### Description and Justification

I ask for the possibility of being able to lock the navigation cube, useful when you are in a 2D sketcher environment and you use the right mouse button to exit the functions, inadvertently you can move the front view by tilting the sketch.

FEA-eng, 2024-09-16:
Why is it a backport request? Can you resubmit this as a standard issue?

Roberto8303, 2024-09-16:
Hi, can you paste me the link to the standard problem?

FEA-eng, 2024-09-16:
> Hi, can you paste me the link to the standard problem?

Sure, try this: https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage&projects=&template=PROBLEM_REPORT.yml

It's the second option called "Report a Problem", under "Request a Backport" that you apparently clicked (the order can be confusing btw.).

closed 2024-09-16 by Roy-043