FreeCAD/FreeCAD #16564: Bug in the Mesurments dialoge. Measurments do not disappear when reset is used.

byteofsoren, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Use the button Measure or use Tools -> Measure, then measure any thing in the model.
After you have created the measure, the fifth item in [std Measure](https://wiki.freecad.org/Std_Measure/en) say that you should click the reset button to delete the measurements but, the measurements don't disappear. 
This has not bean a problem before this release as far as I know.

For now the only work around is to save and close the project and re-open it. 

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.2.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-09-16:
I can confirm it. @maxwxyz IMO this should be a blocker, maybe even RC blocker.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

maxwxyz, 2024-09-16:
@hlorus

Murmele, 2024-09-16:
https://github.com/FreeCAD/FreeCAD/blob/b6599c5315ba557371dbdacca0b9cf35f5b7f7f2/src/Mod/Measure/Gui/TaskMeasure.cpp#L407-L419

Why only the members are set to nullptr, but no "removeObject()" is done? This is before my PR and after so it should be the problem, but I did not understand why it even worked before?

Did I introduce with: https://github.com/FreeCAD/FreeCAD/pull/16379 ?

luzpaz, 2024-09-16:
@benj5378 may involve your previous commit :point_up_2:

benj5378, 2024-09-16:
I will look at this today or tomorrow.

benj5378, 2024-09-16:
I looked at this, and this should be reverted (as it cannot be fixed): a7831eee4bd392dfa77f21f662466a9e72f21864

Murmele, 2024-09-17:
> I looked at this, and this should be reverted (as it cannot be fixed): [a7831ee](https://github.com/FreeCAD/FreeCAD/commit/a7831eee4bd392dfa77f21f662466a9e72f21864)

Didn't work to add a removeObject() to the reset function?

benj5378, 2024-09-17:
> > I looked at this, and this should be reverted (as it cannot be fixed): [a7831ee](https://github.com/FreeCAD/FreeCAD/commit/a7831eee4bd392dfa77f21f662466a9e72f21864)
> 
> Didn't work to add a removeObject() to the reset function?

No, issue is, that explicit mode etc. should be reset as well. So to avoid code duplication, it is better reverting this.

benj5378, 2024-09-17:
> You mean this does not work? [benj5378@8090feb](https://github.com/benj5378/FreeCAD/commit/8090feb83524717b9f33974d26e515e5771fa6c2)

The specific commit, no, because it would clear selection when there are no measuretype. Let's imagine, that you select one vertex; there is no measurement (measuretype) for this, so the reset will be triggered and clear the selection. That means, that one would never be able to select two vertices to get a distance measurement.

closed 2024-09-18 by chennes

NomAnor, 2024-09-25:
I think this issue is still present in master. I created a Body wit a Pad and then I activate the measure tool, click on a face and it shows the area in the 3D view. When I click reset, the area is still displayed. Closing the tool or reopening it doesn't help. When I click on another face, the previous measurement is removed and the new area is shown. This seems to be the case for all measurement modes.

```
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38513 (Git)
Build type: Release
Branch: main
Hash: 7c797b5a10c19ae55e317c4ff9a02823a4b4dc0b
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.5.27
```

benj5378, 2024-09-25:
@NomAnor, from your about info, I can tell, that you are not using the latest version of FreeCAD. In fact, it's prior to both the fix and the issue introduction. Can you please try the newest version from main branch?

NomAnor, 2024-09-25:
This is a build of the current main branch on commit 54c278c123. The only difference to the cmake release preset is `FREECAD_QT_VERSION=6` and `CMAKE_CXX_FLAGS=-Wno-deprecated-declarations`.
The same problem exists for me in `FreeCAD_weekly-builds-38827-conda-Linux-x86_64-py311.AppImage`.
I also started FreeCAD with empty config files (`-u user.cfg -s system.cfg`).

benj5378, 2024-09-25:
Bug introduced with af9097ce875d1378fd6487bc5ffedb35bd148274. @hlorus, didn't you mean to rename all variables and remove `_mMeasureType` from the header file?

Murmele, 2024-09-25:
I tested recently on windows and there it worked fine.

hlorus, 2024-09-26:
@benj5378  Yes, looks like there was an unwanted change when merging.

PR #16588 (2024-09-18): Revert "[Measure] Fix redundant variable assignment"
Fixes #16564 introduced in a7831ee
commit a6975c8c (2024-09-18): Revert "[Measure] Fix redundant variable assignment"

Fixes #16564. This reverts commit a7831eee4bd392dfa77f21f662466a9e72f21864.
commit 89b118e1 (2024-09-18): Revert "[Measure] Fix redundant variable assignment"

Fixes #16564. This reverts commit a7831eee4bd392dfa77f21f662466a9e72f21864.